ArduPilot/ardupilot #22003: CubeOrangePlus: make bdshot firmware

rmackay9, 2022-10-19:
I have a request from a user in Japan (received offline) requesting that we add a CubeOrangePlus-bdshot firmware.  This makes sense because we already have a CubeOrange-bdshot and the CubeOrangePlus is its replacement

FYI @andyp1per @bugobliterator

bugobliterator, 2022-10-19:
> I have a request from a user in Japan (received offline) requesting that we add a CubeOrangePlus-bdshot firmware. This makes sense because we already have a CubeOrange-bdshot and the CubeOrangePlus is its replacement
> 
> FYI @andyp1per @bugobliterator

I plan on enabling bdshot by default on standard build

rmackay9, 2022-10-20:
@bugobliterator,

OK, thanks for the reply.  I think that sounds like a good idea long term but is that something you'd be aiming to back port to 4.3?  If "not" then I think we should make CubeOrangePlus consistent with CubeOrange and provide a bdshot version.

bugobliterator, 2022-10-21:
@rmackay9 Here is the PR for this https://github.com/ArduPilot/ardupilot/pull/22029 .And test firmware https://firmware.cubepilot.org/PublicTestFW/arducopter_cubeorange_bdshot.apj . If you have user(s) to test this that'd be great.

rmackay9, 2022-10-21:
@bugobliterator, thanks very much!  I've asked Wagata-san to help test (he was the one who asked for this change).  Thanks!