ArduPilot/ardupilot #22009: Toggling safety switch state by MP can kill all motor outputs.

ROSStargh, 2022-10-20:
How to reproduce the problem(Tested Copter v4.2):
    1. Take off the aircraft.
    2. Press "Ctrl+F" in MP.
    3. Press the button "Toggle Saftey Switch".
    4. The aircraft crashes.

If the physical safety switch is shorted during a flight, nothing bad will happen as long as the parameter BRD_SAFETYOPTION is set properly. However, using MP to toggle safety switch behaves differently. I think it is confusing and can be dangerous for someone who needs the function.

The safety switch should not be allowed to be activated in flight via mavlink commands.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2022-10-20:
See https://github.com/ArduPilot/ardupilot/issues/21880#issuecomment-1267454330

ROSStargh, 2022-10-20:
#7985
#8156

Changes were made to prevent physical safety switch having chances to kill motors in flight, but why people still let software switch be able to do that? It is a big confusion. 

The parameter BRD_SAFETYOPTION should decide whether the user want the safety switch to be functioning in flight or not, but the mavlink command can ignore the setting. I got a crash because I was not aware of how it worked. Very frustrating. If there really is a need to kill motors in flight, I think the function "motor emergency stop" can do it.

By the way, when a multipoint network is used to control multiple drones, the possibility of killing motors by mistake increases. Because the mavlink command is sent to all drones connected when pressing the "Toggle Saftey Switch" buttion. Mission Planner is designed to do so.

IamPete1, 2022-10-20:
The assumption is that no-one is going to be able to press the physical button when the vehicle is in flight, so any press received must not be genuine. 

As it it quite possible to trigger the safety button remotely when the vehicle is in flight ArduPilot must assume that the trigger is genuine and not second guess the operator. We could add a new option to ignore the safety button command, but we cannot make that the default behavior.

ROSStargh, 2022-10-20:
Adding an option to ignore the command is not a bad idea. Having an option would make the pilot clear about how the drone reacts to the command. And it should be clarified in the document that the safety switch has different behavior depends on whether it is triggered by a mavlink command or not.

rmackay9, 2022-10-25:
Related PR https://github.com/ArduPilot/ardupilot/pull/22051
Related MP issue https://github.com/ArduPilot/MissionPlanner/issues/2987