ArduPilot/ardupilot #22010: Invensense V3 no gyro AAF/NF

fvantienen, 2022-10-20:
https://github.com/ArduPilot/ardupilot/blob/3521090dd5d5a5f5345d3cf437d9657f7b00630e/libraries/AP_InertialSensor/AP_InertialSensor_Invensensev3.cpp#L581

Isn't this driver missing the configuration for `INV3REG_GYRO_CONFIG_STATIC2`, to unset the `GYRO_AAF_DIS` and/or `GYRO_NF_DIS`? Currently only the AAF is enable on the accelerometer and not the gyro.

andyp1per, 2022-10-20:
No, the bit layouts for gyro and accel are different

fvantienen, 2022-10-20:
[image]
As you can see this is not correct for the **ICM-40605** and the **ICM-40609D**, there the register is reset to 0x3 which disables both the AAF and NF. I'm not sure that was intended?

andyp1per, 2022-10-20:
You are right the defaults are different on these two boards vs the 42688

andyp1per, 2022-10-20:
I have created a PR to fix this.

PR #22017 (2022-10-25): Ensure that hardware AAF and notch filter are enabled on 42605 and 42609
Fixes #22010
closed 2022-10-25 by tridge