ArduPilot/ardupilot #22011: RCinXScaled option is restricted by Ardu PWM MIN/MAX values

Sibi4711, 2022-10-20:
Hi,

I'm using stable 4.3 on a plane and have a PAN Servo on Channel 8 which is able to travel PWM 680 - 2235.

So this is how it is set up:

RC8_DZ,0
RC8_MAX,2235
RC8_MIN,680
RC8_OPTION,0
RC8_REVERSED,0
RC8_TRIM,1500

SERVO8_FUNCTION,147
SERVO8_MAX,2235
SERVO8_MIN,680
SERVO8_REVERSED,0
SERVO8_TRIM,1500

The OpenTx radio sends max travels from PWM 680 - 2235. However, it seems that the servo is still restricted by the min and max allowed values in Ardu which are 800 - 2200. Can we lift that limit so I don't have work with a HW Servo extender? I have also tried RCPassthru (1) and RCin8 (58) with the same result.

Thanks!

IamPete1, 2022-10-20:
The param limits are not hard coded. Use the full list to set to the larger range.

closed 2022-10-20 by IamPete1

IamPete1, 2022-10-20:
I did find a bug when poking about at this stuff. I suspect your problem is actually on the RC input side. I would set the transmitter back to its normal output range so your `RC8_MAX` is about 2000 and your `RC8_MIN` about 1000. Then the scaling on the output function should take care of the extra range.

Sibi4711, 2022-10-20:
Thanks Pete!
Your workaround of RC8_MAX 2000 and RC8_MIN 1000 works! 
best Seb

Target0815, 2022-10-21:
No, it does not work ...

Channel 10 in the transmitter, RCIN10Scaled, output to #8 in the servo output, RC10 is set to Default

[image]

The range can be set as desired, but only 800-2200 is output.

[image]
[image]

IamPete1, 2022-10-21:
@Target0815 Use the full parameter list to set outside the recommended range.

Target0815, 2022-10-21:
@IamPete1 Yes, I can confirm. However, it does not work cleanly. If I set SERVO8_MIN = 650 for example, it will output 727 or 922 us for a short time before the value drops to 650 us. Whether this is due to the display in ToolkitRC or to the servo output I cannot verify at the moment.

jreise-d, 2022-10-21:
I can confirm this, tested with oscilloscope. If I set SERVOx_MIN to a value < 800 (650 for example), after a "out of range" warning this value is stored. But the output on the servo pin stays at 800µs. I think in ardupilot servo output is generally limited to 800-2200 by design.

Target0815, 2022-10-21:
No, this is not correct or it may depend on the firmware version. In my case I can set _MIN to 650 in the FPL as written, save the parameters and these 650 are then also displayed in the servo output and output to the port.

jreise-d, 2022-10-21:
Tested with 4.3.0 stable. Set and save MIN as 650, output is 800.

IamPete1, 2022-10-21:
@jreise-d which flight controller and which output pin?

jreise-d, 2022-10-21:
It's a F765wse.

SERVO7_FUNCTION,146
SERVO7_MAX,2000
SERVO7_MIN,650
SERVO7_REVERSED,0
SERVO7_TRIM,1500

RC7_DZ,0
RC7_MAX,2000
RC7_MIN,1000
RC7_OPTION,0
RC7_REVERSED,0
RC7_TRIM,1500

output pin is 800µs, tested with osci

IamPete1, 2022-10-21:
@jreise-d it could be a issue on the RC input  side. Try `SERVO7_FUNCTION` 134 "Servo min", that will decouple the servo output from the RC input.

jreise-d, 2022-10-21:
same. But I found a solution, very strange:

In the fpl I set SERVO7_MIN to 1000 -> servo pin instantly outputs 1000µs (oszi)
set it (and of course write params) to 800 -> output 800µs
set it to 650 -> output 800µs
set to 900 -> output 900µs 
...
and now the strange thing:
set to 650 -> output 800µs
set e.g. TRIM to 1501 -> outputs 650µs !!
it's reproducable, any second changed parameter results in the correct pwm output
strange ...

Sibi4711, 2022-10-21:
@jreise-d 
so you set SERVO7_TRIM,1501 and 650µs is put out?!

jreise-d, 2022-10-21:
yes

jreise-d, 2022-10-21:
Omg, it's my stupid mistake!
After set to 650 at the "value out of range" message I click yes and then forgot the second click "write" 

sorry