ArduPilot/ardupilot #22050: source code building error

hackfaq, 2022-10-24:
Hello, I cloned repository, 
build a docker image: ```sudo docker build . -t ardupilot:latest```
run it with command: ```sudo docker run --rm -it -v `pwd`:/ardupilot ardupilot:latest bash```

then run in container 
```
./waf configure --board Pixhawk6X
./waf copter
```

I get build errors:

```Padded 20 bytes for bootloader.bin to 17888
Embedding file bootloader.bin:/ardupilot/Tools/bootloaders/Pixhawk6X_bl.bin
Embedding file hwdef.dat:/ardupilot/build/Pixhawk6X/hw.dat
Embedding file io_firmware.bin:Tools/IO_Firmware/iofirmware_lowpolh.bin
Embedding file param/CUAV_V6X_defaults.parm:/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/Pixhawk6X/param/CUAV_V6X_defaults.parm
[ 9/10] Linking build/Pixhawk6X/modules/ChibiOS/libch.a
make: *** No rule to make target 'CrashCatcher_armv7m.S', needed by 'modules/ChibiOS/obj/CrashCatcher_armv7m.o'.  Stop.
make: *** No rule to make target 'CrashCatcher_armv7m.S', needed by 'modules/ChibiOS/obj/CrashCatcher_armv7m.o'.  Stop.

Waf: Leaving directory `/ardupilot/build/Pixhawk6X'
Build failed
 -> task in 'ChibiOS_lib' failed (exit status 2): 
	{task 139806106570192: ChibiOS_lib hwdef.h,hw.dat,include```

IamPete1, 2022-10-24:
Have you updated submodules? `./Tools/gittools/submodule-sync.sh`

rmackay9, 2022-10-25:
I think this is more of a developer support question so I'll close this issue if that's OK.  It looks like the solution is likely above and we will see replies.  Sorry to be strict but if we're not our to-do list turns into a support forum.

I'm actually not sure if we have a developer discuss section but also asking on Discord would work.

closed 2022-10-25 by rmackay9

nanshan-dark-cat, 2024-07-04:
> 你更新了子模块吗？`./Tools/gittools/submodule-sync.sh`

it works