ArduPilot/ardupilot #22095: Storm32 mavlink gimbal control with AP 4.3

Tangoviktor, 2022-11-01:
## Bug report
Mavlink gimbal controls in auto and guided modes relating frame vs earth and limit angles not working

**Issue details**
unexpected behaviour regarding storm32 mavlink gimbal.

_Please describe the problem_

When flying in auto or guided mode Gimbal stays in "earth" frame with 2 axis gimbal. You can fly forward, turn the yaw in either direction and when you tilt the gimbal, it automatically turns the yaw axis back to the direction of travel. It happened while in moving and stationary also. I changed storm32 settings to HoldHoldHold instead of HoldholdPan and it helped while being stationary, but moving is still an issue.

Second issue is regarding the roll limit axis - they do not work. I can change the value from 1-2000 and it does nothing to limit the roll travel. Same with pitch. Can only change the pitch axis travel from storm gui.

Also - would be good to have the possibility of choosing the earth vs frame lock withouth the RC commands - because we do not use any kind of RC remotes, only gamepads.

**Version**
_What version was the issue encountered with_

4.3.0

**Platform**

[x ] Copter

**Airframe type**
X frame

**Hardware type**
Cube orange

Tangoviktor, 2022-11-01:
Hello, 

here is the log file. This was done with the previous stable build of ardupilot. We are now going to test 4.3 today.
[2022-10-27 15-16-37.zip](https://github.com/ArduPilot/ardupilot/files/9909516/2022-10-27.15-16-37.zip)

Tangoviktor, 2022-11-01:
Just a recap  - the issue was not present in guided mode, but remained in auto mode. Did not find a place in mp to change the aux value. Will try to update it.

rmackay9, 2022-11-03:
@Tangoviktor,

Re MP's Aux Function tab, it should be visible if you update to MP beta (use the Help screen's "Check for Beta Updates" button).  then right-mouse-button click on any one of the tabs and select, "Customise" then click the checkbox next to "tabAuxFunction".

OK, so it sounds like you're saying the issue is present in 4.2.3 as well which is good news from my point of view actually because it means we haven't made the situation worse with 4.3.

By the way, you said you're using a 2 axis gimbal.  Those two axis are pitch and yaw (aka tilt and pan)?  You also talk about roll so I'm slightly confused about the physical setup.

I wonder if you could also confirm which SToRM32 gimbal you're using?  As it says on [the wiki](https://ardupilot.org/copter/docs/common-storm32-gimbal.html) there are two possibilities

- SToRM32 gimbals with I2C setups and firmware v0.96 should use the SToRM32 Serial driver
- SToRM32 NT gimbals running firmware above v0.96 should use the SToRM32 MAVLink driver

FYI @olliw42

olliw42, 2022-11-03:
@rmackay9 
many thx for the FYI

> By the way, it would be best to [report this in the support forums](https://discuss.ardupilot.org/c/arducopter/6) first and then once it is confirmed to be a bug we can raise an issue here.

to be fair, he was first asking in the STorM32 rcgroups thread https://www.rcgroups.com/forums/showthread.php?2055844-STorM32-BGC-32-bit-3-axis-brushless-gimbal-controller/page283&perpage=50#post50006567, but didn't got any response
(including not from me)(I just couldn't make sense of the report and thus no idea popped up in my head immediately, I was quite busy so it slipped through, sorry, to both of you)

I guess one would first want to understand if this is a STorM32 issue or ArduPilot issue. STorM32 is somewhat weak in 2-axis configs, especially if it would be tilt-pan (it's a 3-axis controller :D !!), so I wouldn't rule out it's a STorM32 thing.

I fully agree with @rmackay9 that we just need much more basic info
- pitch-roll or pitch-yaw (tilt-pan) gimbal?
- STorM32 firmware version?
- STorM32 parameters (the STorM32 gui offers a share parameters feature, pl use it)
- usually I start to understand a strange issue only well enough if I see e.g. a demo video showing it

I guess it would be inappropriate to have the discussion, if it is a STorM32 thing or not, here in as issue. So, @Tangoviktor , I suggest that you report more info in the STorM32 rcgroups thread, sufficient info for me to understand what you describe is going on, and I will treat you there and we will sort out if it's a STorM32 thing. If it is identified a ArduPilot thing you may go back here. 

Does this make sense?

Olli

Tangoviktor, 2022-11-03:
Hi guys! 

Sorry for not clarifying before. I am using pitch-roll configuration with storm32 v2.53 fw verison. I tried v2.59e also, but with that one i did not get mavlink communication working with the drone. I will add more information about the storm32 to the rcgroups also. PS! Having roll and pitch limit angles for the gimbal stabilization would be great for storm32. 

More about the issue - 4.2.3 fw in guided mode the gimbal always had earth reference. I could yaw the drone and when I tilted the gimbal (pitch axis) it automatically turned the drone back to the heading. It changed when in storm32 gui I used hold/hold/hold instead of hold/hold/pan. Then i could move the gimbal in guided mode while in hover, but when moving to the next waypoint it showed same problem
Same was in auto mode - "yaw axis" of the gimbal (although it is the yaw of drone) stayed the original heading.  So in essence in guided and auto mode,  changing the position in pitch axis of the gimbal triggered violent yaw movement of the drone.

4.3 - in guided mode, mnt_yaw_max/min=0, storm32 in hold/hold/hold everything works as intended - gimbal stays in frame reference. Where i point the yaw i can tilt the gimbal up and down while moving and in hover.
But, the issue still remains in auto mode. Gimbal always has earth reference from the initial heading. Whenever I look up and down with the gimbal, it drone will yaw back to the initial heading that it was going to.
I will try to change the issue with MP beta version.

olliw42, 2022-11-03:
> I am using pitch-roll configuration

thx

> storm32 v2.53 fw verison. 

so it is a NT setup. very good.

> I tried v2.59e also, but with that one i did not get mavlink communication working with the drone.

strange, v2.59e should be better. I kindly ask you to discuss this with me in the rcgroups thread, I am eager to sort this out (I take some proudness in STorM32's mavlink capability :))

be prepared that I will ask you to use the latest firmware version, v2.62i. 

> More about the issue - 4.2.3 fw in guided mode the gimbal always had earth reference. I could yaw the drone and when I tilted the gimbal (pitch axis) it automatically turned the drone back to the [...]

ok, so I guess we can conclude that this must be an ArduPilot issue. My rational: Since it's a pitch-roll 2 axis gimbal, "yaw" always only can refer to the flight controller. AFAIK, the STorM32 has no means to affect the flight controller's orientation/attitude. Ergo, whatever happens with "yaw" is ArduPilot specific.

my first guess, although purely speculative is, that there is some inconsistency in the mavlink messages send to the gimbal (i.e., the gimbal does exactly what it is said via the messages it should do, but that some info in the messages is not as it should be). Ergo, I guess you guys want to carefully inspect the stream of mavlink messages which are sent to the gimbal.

it is super strange btw that you see a difference between "hold/hold/hold" and "hold/hold/pan", since for a pitch-roll 2 axis gimbal it really should not matter. I'd like to discuss this with you and sort out in the STorM32 rcgropups thread, not here.

I guess, we all have now a plan for what our respective jobs are LOL

rmackay9, 2022-11-03:
@Tangoviktor,

I have a hard time understanding how changing the roll and pitch of the gimbal could possibly change the yaw of the vehicle.  Can I ask, how are you controlling the roll, pitch and/or yaw of the gimbal?  the MNT_RC_IN_xxx parameters are all zero so I guess you're not using the transmitter.

Are you using MP's Payload Control window?  This seems unlikely because almost nobody knows about this tab.
[image]

Anyway, if you have answers for these questions great, in any case, maybe I'll leave it to you and olliw42 to discuss in the other forum and then report back.

Thanks in any case for testing AP 4.3's gimbal control.

Tangoviktor, 2022-11-03:
> I have a hard time understanding how changing the roll and pitch of the gimbal could possibly change the yaw of the vehicle. Can I ask, how are you controlling the roll, pitch and/or yaw of the gimbal? the MNT_RC_IN_xxx parameters are all zero so I guess you're not using the transmitter.

We are using a ps4 gamepad with it together with QGC. I have used the payload control in mp also and it moves the gimbal in pitch and roll, but I have yet to tested if the payload control sliders change anything in the same flight conditions as described before. One thing I can say is that on a different drone where we have a basecam mavlink controller the same issue does not occur and i think it is because - the manual it says 

" _NOTE: By default, YAW angle is considered relative to frame (legacy behavior in ArduPilot FC), so we add
YAW angle of the frame as received in the message ATTITUDE. But it can be changed by setting the flag
“YAW angle is absolute” in the GUI_." 

But with storm32 it seems that in guided mode when mnt_yaw=0 it keeps it as a frame reference. 

Also note mentioning - storm32 itself is set to send mount_status mavlink. Maybe it would make a difference when using attitude/mountorient instead of mount status?

Tangoviktor, 2022-11-03:
> that's confusing what you originally said is that TILTING the gimbal has an effect on the YAW behavior, or vice versa while what you suggest as explanation ONLY relates to YAW

Yes, the only issue is with YAW. it only happens when using mavlink commands "gimbal up/down" through QGC+joystick in guided and auto mode (4.2.3) and in 4.3 only in auto mode. 
> 
> there might be indeed a difference with respect to yaw, since STorM32 indeed interprets the yaw angles as relative to the vehicle frame. This was so however since "ever", and I think ArduPilot worked along this line since "ever" too.
> 
> I have a growing suspicion that the issue is that when a "tilt the gimbal" message is sent from ground to the STorM32, it is sent with yaw = 0, i.e., with "align the gimbal forward with respect to the vehicle". If so, there is actually little to nothing STorM32 can do about it (without using STorM32Link, which ArduPilot does not support). But again, this would be then the behavior as "ever", I'm not sure what has happened in between for it to turn up only now.

it only happens in tilt commands yes. I will try to do the same thing tomorrow with MP payload controls, both for roll and pitch axis, just to see if ther is a difference.
> 
> maybe all one needs to do is to somehow disable has_pan. Doesn't 4.3 allow this to do?

I have disabled pan in 4.3 but it only affected the guided mode, not autp

> 
> > Also note mentioning - storm32 itself is set to send mount_status mavlink. Maybe it would make a difference when using attitude/mountorient instead of mount status?
> 
> no. This are just different messages send from the gimbal to the world, but none of them are really digested by ArduPilot. I also think QGC has still the bug that it would get confused by sending attitude. MOUNT_STATUS is the right thing for you.

rmackay9, 2022-11-04:
Txs for the info.  I think we are edging closer to the cause and solution.

So where is this QGC "gimbal up/down" control screen?  How do I get to it after starting QGC?  Coincidentally I had a look at QGC the other day looking for an equivalent to MP's Payload Control screen and couldn't find any.

As @olliw42 says, the gimbal's yaw control can be disabled in AP by setting both MNT1_YAW_MIN and MNT1_YAW_MAX to zero.

@Tangoviktor sorry to be a bit strict, but in order to get to the solution it would help if, when you write "yaw", you could specify each time if you mean **vehicle** yaw or **gimbal** yaw.  I think in general you're talking about vehicle yaw but you're mixing them up quite often which adds confusion.

rmackay9, 2022-11-04:
I had a peek in the logs and we can see the [MOUNT_CONTROL messages](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L1162) that the ground station is sending and we can see the Pitch angle is provided (these are in centi-degrees) but roll and yaw are zero.

[image]

When AP receives a mount-control message it will consume all three axis.  If it thinks the gimbal is 3-axis it will leave it to the gimbal to implement the yaw angle but with a 2-axis gimbal it will yaw the vehicle.  One difficulty with the MOUNT_CONTROL message is that it doesn't specify the frame (e.g. should yaw be earth-frame or body-frame?).  The odd outcome of this is that when using a 3-axis gimbal, yaw of "0" means point forward (e.g. body frame) while with a 2-axis gimbal, yaw of "0" will lead to the vehicle pointing North (e.g. earth-frame).  This is totally crazy of course but I can't think of how to fix this and so instead I just recommend that we change the ground stations to use the new [GIMBAL_MANAGER_PITCHYAW command](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L1514) which has a frame field.

Tangoviktor, 2022-11-04:
> Txs for the info. I think we are edging closer to the cause and solution.
> 
Thanks, I feel like it too.

> So where is this QGC "gimbal up/down" control screen? How do I get to it after starting QGC? Coincidentally I had a look at QGC the other day looking for an equivalent to MP's Payload Control screen and couldn't find any.
> 

The QGC gimbal up/down controls can be seen when you are connected to the autopilot, connect a usb gamepad, and then under the "joystick" tab
[image]

[image]

> As @olliw42 says, the gimbal's yaw control can be disabled in AP by setting both MNT1_YAW_MIN and MNT1_YAW_MAX to zero.
> 

I have mentioned this before, that the issue started for me I think in 4.2.0 where i first experienced the error. In 4.3 I could set the mnt_yaw_min/max to zero and it helped, but only in **guided mode**. I will try to make a video today - a screen recording of what happens with both auto and guided mode.
 
> @Tangoviktor sorry to be a bit strict, but in order to get to the solution it would help if, when you write "yaw", you could specify each time if you mean **vehicle** yaw or **gimbal** yaw. I think [...]

Yes, sorry for that. In general, in this thread I am always referring to vehicles yaw because of my 2 gimbal setup.   
> By the way, the WP_YAW_BEHAVIOR parameters affects how the vehicle's yaw is changed during Auto mode. Currently it is 1 (face next waypoint) which is generally a good choice. 0 will mean the yaw is never changed even as the vehicle passes through waypoints.

Yes, we have always had it 1, so that it would face next WPs.

Tangoviktor, 2022-11-04:
> I had a peek in the logs and we can see the [MOUNT_CONTROL messages](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L1162) that the ground station is sending and we can see the Pitch angle is provided (these are in centi-degrees) but roll and yaw are zero.
> 

If this i from my log, then yes, this is what it is supposed to do. Well, this is what i tell it to do. Sorry I am not that proficient in MP, but if you could  compare the vehicle yaw with the time of when the command has been given in **guided** and **auto** mode, I think you can see where the drone has done a violent 180 degree turn. Specifically in auto mode, in the last leg of the mission, when coming back to RTL

> [image]
>
> When AP receives a mount-control message it will consume all three axis. If it thinks the gimbal is 3-axis it will leave it to the gimbal to implement the yaw angle but with a 2-axis gimbal [...]

But if we can use RC commands to switch between earth and body frame, cant we have an additional parameter to lock the gimbal? Such as there could be a Mnt_axis parameter that could define either a 1, 2 or 3 axis gimbal and by default 
 1 and 2 axis gimbals would be body frame and only with the 3 axis gimbal mnt_yaw_min/max parameter would be present.

olliw42, 2022-11-04:
@rmackay9 
I disagree with your analysis
IMHO the root cause of the issue(s) is that one and the same message is used for both controlling the gimbal and vehicle yaw. Over the years this has cretaed numerous issues with on off bugs and issues. And it's incorrect that they would be fully resolved by just one relative/absolute flag.
IMHO with your intended use of GIMBAL_MANAGER_PITCHYAW you will write in stone a missunderstanding, will shatter the standard concept into pieces, will produce more disgression and distinction instead of making the various systems better interoperable. Also, sadly, neither in the mavlink nor ArduPilot communities the limitations imposed by encoder-less gimbal designs seems to be understood.
Sadly, I'm fighting against windmills here. 
Since STorM32 will stick in spirit with the standard concept, you march towards a point in time where you will conclude that you want to drop any STorM32 gimbal drivers. I'm not saying this is necessarily bad, but one should be clear and honest about consequences.

rmackay9, 2022-11-07:
@olliw42,

I don't understand the 2nd part but I do agree that a core problem is that MOUNT_CONTROL can control the attitude of the vehicle.  I actually thought of removing this ability as part of my recent work on the gimbals but stopped because another user was relying on it when the MOUNT_CONTROL command is used in Auto mode (I think it was @rrr6399).  If I remove this ability he would need to enter two commands in auto (MOUNT_CONTROL and DO_CONDITION_YAW) or perhaps he could use the new DO_GIMBAL_MANAGER_PITCHYAW.

Another solution to this problem is to change QGC to use DO_GIMBAL_MANAGER_PITCHYAW instead of MOUNT_CONTROL.  MOUNT_CONTROL is really not a good message for us to be using anymore.

Tangoviktor, 2022-11-07:
I can try that. At the moment mnt1_yaw_min and max =0. I can try to put it like 1 and 2, maybe it helps the issue that im having. Here are the videos also. 
@olliw42 Here you can see that in RTL when moving the gimbal in pitch axis, the drone clearly yaws. In guided mode it does not do that anymore, but only in auto and RTL mode. 

https://user-images.githubusercontent.com/10380859/200290156-e1cfae82-af95-428a-be67-af54f068034c.mp4

rmackay9, 2022-11-07:
@Tangoviktor,

Great, thanks.  By the way, the "Yaw Imbalance" messages are because there are persistent differences in the speed of the CW vs CCW motors.  Maybe one of the motors is slightly tilted or something.  I'm not totally sure and I'm also not saying that it's the cause of the changes in yaw you're seeing.  Just pointing out the cause of this message.
[image]

Tangoviktor, 2022-11-07:
Yea, I know the issue. One of the motors is tilted slightly because of the way how the arm was assembled, but for the point being it does not matter. It is a testing rig anyway.  But thanks :)

rmackay9, 2022-11-26:
I think I'll close this issue if that's OK because I think we've confirmed this is a short-coming in QGC's button feature.

closed 2022-11-26 by rmackay9

shivamydv2001, 2023-10-26:
how to control STorM32 gimbal controller(board 1.32) uisng mavlink and I am using Raspberry pi4