ArduPilot/ardupilot #22173: Rover: allow slower steering response in Manual mode

rmackay9, 2022-11-10:
Users sometimes find that manual modes steering response is too fast especially on small skid-steering vehicles.  This is because the pilot directly controls the steering giving them access to the maximum turn rate of the vehicle (other modes use the turn rate controller and the maximum rate can be limited with ACRO_TURN_RATE or ATC_STR_RAT_MAX).

We should add a new parameter called MANUAL_TURN_SCALE (?) that allows the user to provide a scaling value (>0 and <=1) to be applied to the pilot's steering input (see desired_steering in mode_manual.cpp) before being passed to the motors library.  This parameter should default to 0 and when 0 it should have no effect.

This enhancement request comes from [this discussion](https://discuss.ardupilot.org/t/rover-4-3-0-beta4-has-been-released/92333/13).

IamPete1, 2022-11-10:
I think the balance bot case is a bit weak its not really true manual. However, we don't seem to have any option for manual mode stick expo as we have on plane (manual) and copter (acro). Maybe some stick expo would allow the user to keep onto of the higher rate more easily.

rmackay9, 2022-11-10:
@IamPete1, txs for the expo idea.  This request is for the steering response to be reduced, on a balance bot the steering is just like a regular skid-steering rover.  Throttle control is different because it goes through the pitch controller.

rmackay9, 2022-11-10:
The user has confirmed this fixes their problem and I've merged the PR to master so closing, thanks!

closed 2022-11-10 by rmackay9