ArduPilot/ardupilot #22183: Rover: scurve vs L1 performance

xianglunkai, 2022-11-11:
I found that the scurve navigation is indeed much better than L1 navigation in handling turns, but it is slightly less accurate in straight line tracking.

I think the problem is that there are too many closed loops involved in scurve! The position of the scurve output is used to calculate the velocity vector through feedback, and then the velocity feedback is used to calculate the acceleration vector, which is converted into linear velocity and angular velocity instructions.

L1 directly calculates the lateral acceleration based on the current position velocity

How to improve the precision of scurve tracking?

rmackay9, 2022-11-14:
@xianglunkai,

Thanks for the report.

One clarification is that the acceleration vector is only used to calculate the lateral acceleration / turn rate.  The desired velocity vector (calculated from the position error) is used to calculate the linear velocity (which means forward-back velocity I think).

xianglunkai, 2022-11-16:
@rmackay9 Txs!  It may be caused by parameter settings.

   According to my observation, it takes longer time for the scurve to return to the route than L1, which may be caused by the low gain of PSC position control(default = 0.2).

  1.  I find that the setting of maximum lateral acceleration, maximum turning radius and maximum longitudinal acceleration has a great impact on the scurve. Is there a better way to set these values reasonably? 
  
 2. const float jerk_max = is_positive(_jerk_max) ? _jerk_max : accel_max; [in AR_WPNAV.cpp]
 I don't quite understand why setting jerk like this, which leads to the fact that jerk is associated with accel, which seems incorrect?

xianglunkai, 2022-11-16:
// get the lateral acceleration limit (in m/s/s).  Returns at least 0.1G or approximately 1 m/s/s
    float get_turn_lat_accel_max() const { return MAX(_turn_lateral_G_max, 0.1f) * GRAVITY_MSS; }

   If the maximum angular velocity of the big boat is only 20 degrees per second and the turning speed is 1 m/s, then the lateral acceleration is less than 0.1 g, is this procedure reasonable?

rmackay9, 2022-11-23:
@xianglunkai,

Do you think you have logs from the same vehicle running S-Curves vs L1?  Leonard Hall and I would like to do a more detailed comparison of performance to see if we can get S-Curves performing better.

xianglunkai, 2022-11-24:
@rmackay9，
Thank you very much for your proposal！
Recently, our ship went out to work. I will send you the comparison logs  as soon as possible.

rmackay9, 2022-12-07:
@xianglunkai,

no huge pressure but looking for to the logs :-)

rmackay9, 2023-09-06:
I think I will go ahead and close this because we never received any logs and from my testing at least I don't think the SCurve straight line performance is worse than L1 if the underlying steering turn rate controller is tuned well.

I'm happy to re-open this if @xianglunkai but I don't see how we can move forward on this.

closed 2023-09-06 by rmackay9