ArduPilot/ardupilot #22190: Solo Gimbal tilt angle limited to 45 degrees in AC 4.3

mtbsteve, 2022-11-12:
## Bug report
With AC 4.3, there is a new parameter setting for RC6_OPTION = 213 (Mount Pitch) introduced and set as default in the parameter file. With the setting to 213, the Solo gimbal tilt is limited from o to approx 45 dregrees. Tilting to 90 degrees is no longer possible.
 
**Issue details**
When I set RC6_OPTION to 0 as in the previous AC versions, the gimbal stalls at about 45 degrees.

_Please describe the problem_

**Version**
_What version was the issue encountered with_
AC 4.3.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad X

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
CubeSolo

**Logs**
_Please provide a link to any relevant logs that show the issue_
My AC4.3 Parameter file:
[https://1drv.ms/u/s!AnKeW8KMoCcygYdRfWAA-EnQjewg8A?e=4wNxF0](url)

rmackay9, 2022-11-15:
@mtbsteve,

I think the issue is the RC6_TRIM.  Could you try setting this to 1500?

mtbsteve, 2022-11-15:
@rmackay9 thanks! With RC6_TRIM set to 1500 the gimbal sweeps now over the full 90 degrees, BUT the travel speed is no longer linear over the course from 0 to 90 degrees sweep angle. 
The first 45 degrees from pointing forward are rushed trough no matter what the preset sweep speed is, and only the remaining 45 degrees to fully pointing downward are traveled at the preset sweep speed. 
I am pretty sure that it was working fine in the 4.3-dev versions I tested in summer.

closed 2022-11-25 by rmackay9

rmackay9, 2022-11-26:
Hi @mtbsteve, I closed this without fully reading your final comments.  I think the change is speed is just related to where the RC6_TRIM value is set.  Perhaps it is not in the middle between RC6_MIN and RC6_MAX.  We did make a lot of changes to the gimbal drivers but I don't think this particular bit of behaviour has changed.  Sorry to ask but if you could go back and re-test with 4.2 to confirm the behaviour is the same that would be a great help.

I'm happy to re-open this issue of course if we still have a problem.

mtbsteve, 2022-11-26:
@rmackay9 thanks for the hint. Since the Solo gimbal travels between RC6_MIN=1000 and RC6_MAX=1520 by default, I had to set RC6_TRIM to 1260 and now everything works fine and this issue can be closed.