ArduPilot/ardupilot #22225: AC 4.3.0 gimbal mountpan bug

andrefreitas97, 2022-11-17:
## Bug report

**Issue details**

When servo function "mountpan" is selected in order to control the pan movement of a servo gimbal, it is verified a problem that when the RC input that controls mountpan exceeds a pwm value of aprox 1700 us, the servo movement direction reverts because it jumps to an input of minimum pwm value. This is, if we move the gimbal pan to the left it works ok, if we move slowly to the right without exceeding 1700 pwm value it works ok, but if we try to move fast to the right exceeding 1700 pwm value it starts moving to the left.

I already verified in radio calibration and the stick movement of the selected RC input for the pan movement is working ok moving from 1051 to 1951 pwm value so the problem is not with the controller.

I also tried moving the "mountpan" function to the servo that controls the tilt of the gimbal in order to see if it was a problem with the servo but the same thing happens so the gimbal servos are ok.

When I go to the servo output page in mission planner i can see the servo output jumping from max to min when I increase the RC input past aprox 1700 us so I am assuming this is a firmware bug. 

**Version**
4.3.0

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Octocopter

**Hardware type**
Orange Cube

rmackay9, 2022-12-19:
Sorry for missing this.  I've added this to our [4.3 issues list](https://github.com/ArduPilot/ardupilot/issues/21711) and I'll have a look.

My guess is the issue is the gimbal is being controlled earth-frame instead of body-frame but in any case I will investigate.

rmackay9, 2022-12-19:
@andrefreitas97,

OK, this is indeed a bug and I've created PR https://github.com/ArduPilot/ardupilot/pull/22442 to resolve this.

Any chance you could give this a try?  I could also create a binary for you if you'd like.

andrefreitas97, 2022-12-19:
Thank you for your attention @rmackay9!

I posted this bug sometime ago and meanwhile I found a workaround. If I changed the MNT1_YAW_MAX value to less than 180 this problem would disappear in RC targetting mode. However, I knew this was not a fix to the problem because if I used MAVLink Targeting and moved the sliders in the Payload Control tab, the problem still occured.

rmackay9, 2022-12-19:
@andrefreitas97,

Great, I was actually going to suggest that workaround.  Anyway, I suspect this will be released in an upcoming "point" release.. probably 4.3.3.

rmackay9, 2023-01-18:
The fix for this has been included in 4.3.3-rc1 which is now in beta testing.  This will be released as the stable version in the coming few days.

I'll go ahead and close this issue now, thanks again for the report!

closed 2023-01-18 by rmackay9