ArduPilot/ardupilot #22228: Here2 GPS support broken by PR 21813

rmackay9, 2022-11-18:
Testing with master on a CubeOrangePlus and older Here2 GPS the vehicle is unable to get a GPS lock.  Instead "GPS:No GPS" is shown on the ground station and the flight code reports, "EKF3 waiting for GPS config".

Through a bisect I've found the cause of the problem is PR https://github.com/ArduPilot/ardupilot/pull/21813

FYI @tridge @bugobliterator

rmackay9, 2022-11-18:
The Here2 GPS can be upgraded by connecting with Mission Planner and then:

- Select Setup >> Optional Hardware >> DroneCAN/UAVCAN
- Push the MAVLink-CAN1 button (if the GPS is connected to the CAN1 port)
- Look for, "com.hex.here" and push the "Menu" button on the right side
-  Menu >> Update
- When asked, "Do you want to search the internet for an update?", press "Yes"
- "Progress fw.bin" will be displayed and the upgrade should start automatically
- Once complete the "SW Version" should be "1.7.3B7DB5CC"

[image]

WickedShell, 2022-11-18:
@rmackay9 Given that we intentionally removed the message and a fix is available, is this something we can mostly just mark as a documentation issue and close?

rmackay9, 2022-11-18:
@WickedShell,

Maybe but we should discuss (at the dev call probably) whether it is really better to force all Here2 users to upgrade their GPSs or just revert the DroneCan change.  If we decide we want them to upgrade then we need to make sure we publicise that.

WickedShell, 2022-11-19:
I'm going to have to miss the next call, but I'd support pushing people forward. The various GNSS DroneCAN messages are pretty bad, and as we have more devices hanging off of CAN these days carrying the support around and having the GPS spam the old message isn't helping us. One less old message to parse is very nice, and it actually saves a surprising amount of flash. From a functionality point of view it's worth pointing out that using `Fix2` at a minimum allows us to report DGPS back instead of just 3D fix.

We could potentially stagger it a bit and remove the AP_Periph side first, and still parse the Fix message for another release, but at a minimum we really need to keep the removal of sending  both `Fix` and `Fix2` from the various sensors on the bus. I think removing the ArduPilot side is also valuable, because it forces vendors to move forward in the interim.

adalta87, 2022-11-19:
> The Here2 GPS can be upgraded by connecting with Mission Planner and then:
> 
> * Select Setup >> Optional Hardware >> DroneCAN/UAVCAN
> * Push the MAVLink-CAN1 button (if the GPS is connected to the CAN1 port)
> * Look for, "com.hex.here" and push the "Menu" button on the right side
> * Menu >> Update
> * When asked, "Do you want to search the internet for an update?", press "Yes"
> * "Progress fw.bin" will be displayed and the upgrade should start automatically
> * Once complete the "SW Version" should be "1.7.3B7DB5CC"
> 
> [image]
> 
> .. and here's a video demonstrating how users can upgrade the GPS https://www.youtube.com/watch?v=yGSL2zgw0oo

why is this specific to the Here2, is this a mistake by cubepilot and hex? why are we going to force people to update, will this possibly break their current running configurations?

rmackay9, 2022-11-19:
@adalta87,

The issue that I hit was specific to the Here2 and it is really due to a change in the DroneCAN protocol over time.  The older Here2 firmware uses the older messages while the newer firmware uses the new messages.  I think a significant number of Here2 GPS users would hit this but we don't know the percentage.

It is also possible that other manufacturer's older GPSs also use the old messages that we are removing support for but we just don't know yet.