ArduPilot/ardupilot #22235: VN-300 only parsing GPS packets (no gyro/accel..)

havi0019, 2022-11-19:
## Bug report

VN-300 connected to CubeOrange on firmware ArduCopter 4-3 or 4-4-dev
The ExternalAHRS is not parsing the gyro/accel data but is only parsing the GPS data.
Put prints in the parsing and I find that it is only getting GPS packets.
VN-300 was reset to factory default and that didn't help. Even tried hard-coding the nmea messages into it with no avail.

**Version**
ArduCopter 4.3.0-dev and 4.4.0-dev

**Platform**
[ x] All
[  ] AntennaTracker
[ ] Copter
[ ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Copter quad-plus

**Hardware type**
CubeOrange w/VN300 on Telem2 
Serial protocol = 36, baud = 115200

havi0019, 2022-11-20:
Was finally able to get resolved!

For others here is what I had to do:
From the VN configurator  - from a stock VN300 setup
1) disable all serial 1/2 messages
2) change baud rate higher than the standard 115200 (bandwidth for the messages is around 90% use so I brought it up)
3) I set NMEA messages that are used -> saved the settings to eeprom

closed 2022-11-20 by IamPete1

rmackay9, 2022-11-20:
It looks to me like we have no documentation on the setup of the VectorNav products.  I've put a note in [our wiki to-do list](https://github.com/ArduPilot/ardupilot_wiki/issues/4274).