ArduPilot/ardupilot #22249: Plane (safety feature) - Execute a WP autoland mission after irretrievable loss of links

robustini, 2022-11-21:
## Feature request

I may remember asking this before, and it would be very helpful.
 Execute a WP autoland mission after irretrievable loss of links could avoid unpleasant crashes if VTOLs are not used.

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

In the case of RTH a fixed wing may already perform an automatic landing, but its path is not fully definable.
For example, I set some landing missions on the runway where I fly, one on the EAST side and the other on the WEST side, depending on the wind direction, And the plane performs both correctly, [here](https://youtu.be/mqiOk_o6O4o?t=313) an example.
So in case of emergency I can perform this, but always if I am connected in telemetry.
Many people who fly FPV do not have a telemetry link available.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

To be able to define a "WP landing mission" that can be activated manually so that it can be tested and aborted if necessary, and that can be executed automatically in the event of a definitive loss of control link.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

We could enter a new EVENT ID in the Mission Planner "Plan" section so that in emergency the Plane would execute all events after that particular EVENT ID. 
In this way one could queue the "emergency mission" to an existing one.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[*] Plane
[  ] Rover
[  ] Submarine

robustini, 2023-04-29:
[This](https://github.com/ArduPilot/ardupilot/pull/23325) is already a good starting point, thanks @WickedShell!

eppravitra, 2023-08-24:
Hi @robustini 
These two pull requests should be able to make what you asked for work. #23333 #23334 
I'm hoping they will get merged soon.

peterbarker, 2024-05-07:
@robustini the two PRs have been merged.  Are we OK to close this issue?

robustini, 2024-06-23:
Yes, closed, thanks!

closed 2024-06-23 by robustini