ArduPilot/ardupilot #22253: Unable to Enumerate Two Identical PX4 Cannodes

AlexKlimaj, 2022-11-21:
## Bug report

**Issue details**

When using two identical PX4 Cannodes, ie dual ARK RTK GPS, Ardupilot is unable to enumerate the nodes. When they are connected one at a time, they are able to enumerate. When connecting two different PX4 cannodes, ie ARK RTK GPS and ARK Flow, the nodes enumerate successfully. The issue is only present with two PX4 Cannodes of the same type. 

The Dronecan GUI and PX4 when acting as the dynamic node ID server are able to successfully enumerate the nodes.

Here is a capture of Ardupilot attempting to enumerate the nodes and failing.
[image]

This is a capture of the dronecan gui acting as the node ID server.
[image]

**Version**
Recent master and the latest release.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Pixhawk 4 Mini and ARK RTK GPS.

tridge, 2022-11-21:
@AlexKlimaj in the captures above, I see all zeros in the unique_id. Is the px4 bootloader properly filling in the hw IDs from the STM32 unique ID?

AlexKlimaj, 2022-11-21:
> @AlexKlimaj in the captures above, I see all zeros in the unique_id. Is the px4 bootloader properly filling in the hw IDs from the STM32 unique ID?

@tridge Looks like it is.

First called here.
https://github.com/PX4/PX4-Autopilot/blob/main/platforms/nuttx/src/canbootloader/uavcan/main.c#L419

Which calls this
https://github.com/AlexKlimaj/PX4-Autopilot/blob/main/boards/ark/can-rtk-gps/src/boot.c#L132

Which calls this that gets the UUID.
https://github.com/AlexKlimaj/PX4-Autopilot/blob/a3ad7106234fee4cc708686b70fcd05ee11920eb/platforms/nuttx/src/canbootloader/arch/stm32/board_identity.c#L46

AlexKlimaj, 2022-11-23:
When each GPS is connected to a different CAN port on the flight controller, they enumerate successfully. It's only when both identical nodes are connected to one CAN port do they fail to enumerate.

AlexKlimaj, 2022-11-24:
This small timing adjustment appears to solves the issue from the PX4 can bootloader side.
https://github.com/PX4/PX4-Autopilot/pull/20663

IamPete1, 2023-02-14:
@AlexKlimaj Sounds like this has been fixed? Can we close?

AlexKlimaj, 2023-02-14:
I fixed it by adjusting some magic timing in the PX4 canbootloader. I was never able to get an ardupilot gdb setup working to debug it on the flight controller side.

https://github.com/PX4/PX4-Autopilot/pull/20663/files

AlexKlimaj, 2023-03-14:
I am receiving reports that two identical ark rtk GPS running ap periph app and bootloader will still occasionally experience this bug.

AverySkyways, 2023-03-14:
Hey all, as Alex said we are encountering this bug on all of our ARK RTK GPS units installed on all of our aircraft. We are using 2 RTK GPS units on each vehicle as a moving baseline configuration. When we get it to boot up correctly, it works flawlessly. However, we are frequently hitting this CAN enumeration boot up error which requires us to reboot the Pixhawk and/or power cycle the aircraft until they can boot correctly (if we don't reboot then both GPS units will forever display "no fix"). This has become very frustrating so we would love to see this fixed. I've been talking with Alex and it seems like we've done everything right on the ARK firmware/software side so it's supposedly an ArduPilot problem.

Can someone from the ArduPilot team look into this in the near future? Thank you in advance!

magicrub, 2023-03-14:
@AverySkyways  @AlexKlimaj Is this happening on the Ardupilot based AP_Periph build using the ArduPilot bootloader?

AlexKlimaj, 2023-03-14:
> @AverySkyways @AlexKlimaj Is this happening on the Ardupilot based AP_Periph build using the ArduPilot bootloader?

Yep, Ap Periph app with AP Periph Bootloader.