ArduPilot/ardupilot #22263: Problem in Copter 4.3.0 with the subsytem led preventing the drone from booting . (luminousbee)

DavideLentini, 2022-11-22:
## Bug report

**Issue details**

Hello all ardupilot developers 😄 , I would like to report a bug that I found while testing the stock Copter-4.3.0 .

First of all I worked together with @ntamas , which will be useful to explain the problem from a second point of view . 

**This bug seems to involve the subsytem led which prevents the drone from booting correctly and preventing us from connecting to it with Mission Planner (both USB and UDP), making any other operation impossible.** 

Just today I tested this problem with the stock _Copter 4.3.0-rc1_ .

To recreate the problem , what I did was to use a **lua script that turns on a number of leds set in the script** ( [lua_led_testing.zip](https://github.com/ArduPilot/ardupilot/files/10067883/lua_led_testing.zip)) .

By setting a number of leds to turn on , in the lua script , **equal to or greater than 44** , turning on the drone the script will start correctly , but **it will be impossible to connect to the drone either from USB or UDP** , and also it seems that the drone does not boot correctly . We have tried several times , **but even from the system led rgb , we can see that the drone did not start correctly** . 

By setting a number of leds to be turned on in the lua script , **less or equal than 44** , turning on the drone the script will start correctly , and **it will be possible to connect to the drone from USB or UDP** ! And the drone will also pass the boot phase correctly . 

**Whatever problem there is, it seems to be triggered when trying to operate with a number of LEDs equal to or greater than 44**

I want to specify that the hwdefs of the drone , have been tested with Copter 4.2.3 and work correctly with all the led subsystem . 

It therefore seems that there is sometype of  bug or problem with Copter 4.3.0 .

I ask for help from the developers to solve this problem as quickly as possible @andyp1per , @rmackay9  . Thank you all :)

**Version**
Copter-4.3.0-rc1

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

andyp1per, 2022-11-22:
This is almost certainly to do with the timeout on dshot pulses - the more LEDs there are the longer the pulse needed to light the LEDs and the more likely this will overrun other users of the DMA channel or output in general. The solution here is re-architecture of the dshot code to have a thread per timer, but that's not going to happen for a while

DavideLentini, 2022-11-22:
> This is almost certainly to do with the timeout on dshot pulses - the more LEDs there are the longer the pulse needed to light the LEDs and the more likely this will overrun other users of the DMA [...]

Yes, I thought it might be a problem like that. We actually had similar problems with 4.2.3, but not so serious that the drone couldn't start! We tried adding the patches that helped us in 4.2.3 , also in 4.3.0 , but they didn't solve the problem of the connection and the drone booting . is it possible that these patches have to be adapted to 4.3.0?

These are the patches :
  path :  _libraries/AP_HAL/RCOutput.h_
      // See WS2812B spec for expected pulse widths
      static constexpr uint32_t NEOP_BIT_WIDTH_TICKS = 20;
      static constexpr uint32_t NEOP_BIT_0_TICKS = 5; //7 to 5 , resolve white led problem
      static constexpr uint32_t NEOP_BIT_1_TICKS = 14;
      // neopixel does not use pulse widths at all
      static constexpr uint32_t PROFI_BIT_0_TICKS = 7;
      
  path : _libraries/AP_HAL_ChibiOS/RCOutput.cpp_
              // timer wrap with ChibiOS timers. Use CH_CFG_ST_TIMEDELTA
            // as minimum. Don't allow for a very long delay (over _dshot_period_us)
            // to prevent bugs in handling timer wrap
            const uint32_t max_delay_us = _dshot_period_us * 5; //resolve partial led on problem
            const uint32_t min_delay_us = 10; // matches our CH_CFG_ST_TIMEDELTA
            wait_us = constrain_uint32(wait_us, min_delay_us, max_delay_us);
            mask = chEvtWaitOneTimeout(group.dshot_event_mask, chTimeUS2I(wait_us));

andyp1per, 2022-11-22:
So the problem is that these changes introduce the bug we were trying to fix - you will be stealing cycles from the dshot output with the possibility of copters falling out of the sky

DavideLentini, 2022-11-22:
> So the problem is that these changes introduce the bug we were trying to fix - you will be stealing cycles from the dshot output with the possibility of copters falling out of the sky

The first test I did, as I wrote, was done on a stock Copter-4.3.0 firmware, so without the modifications I wrote above, and it still resulted in these dshot problems. 
Subsequently I tried adding the patches I wrote above , to make a counter-test with the same result . I understand your concern about the dshot :) 

So if I understand correctly , in the current situation , there is nothing we can do to solve this problem and use Copter-4.3.0 for our drones ?
It would be a great pity, given all the innovations and advantages of Copter 4.3.0 :(

andyp1per, 2022-11-22:
Rough calculation - 1.25us/bit, 24bits/LED, 42 LEDS = 1260us/cycle. So maximum output rate is 793Hz. Default dshot rate is 1Khz. Currently the minimum output rate is 800Hz, so hence why you always timeout. I guess you could try changing the minimum output rate to be 400Hz and see if that helps.

DavideLentini, 2022-11-22:
> Rough calculation - 1.25us/bit, 24bits/LED, 42 LEDS = 1260us/cycle. So maximum output rate is 793Hz. Default dshot rate is 1Khz. Currently the minimum output rate is 800Hz, so hence why you always timeout. I guess you could try changing the minimum output rate to be 400Hz and see if that helps.

Thanks for the advice, the drone has 100 LEDs in total, so I guess it is 100 LEDS = 3000 us/cycle.
This modification of the minimum output rate in which file should be done ? Thanks a lot for the help

andyp1per, 2022-11-22:
Yes.

ntamas, 2022-11-22:
Just to add some more context: in the setup that @DavideLentini is trying, the LEDs are updated at most 50 times per second, so there should be plenty of time available for the DShot pulses - a few of them will be missing every now and then when there's a long LED pulse. But I think that an update rate of 50 Hz for the LEDs should not cause starvation, given that the length of a full pulse for 100 LEDs will be around 3 msec only, right?. (This worked fine with Copter 4.1.x; the workaround that @DavideLentini posted was needed for Copter 4.2.x to make all the 100 LEDs work).

DavideLentini, 2022-11-22:
> Just to add some more context: in the setup that @DavideLentini is trying, the LEDs are updated at most 50 times per second, so there should be plenty of time available for the DShot pulses - a few [...]

Exactly . @ntamas The workaround I have written here , in the case of 4.3.0 , (As you also tested) solved the problem of only 20 leds turning on and the leds only turn on with  white color  BUT it did not solve this problem of the drone not booting .

andyp1per, 2022-11-25:
commit e0adbb978ed13c664bff7cd151cbed77613a14ff
Author: Andy Piper <github@andypiper.com>
Date:   Wed Apr 6 18:44:34 2022 +0100

    AP_HAL_ChibiOS: fix dshot timeout bug where the elapsed pulse is longer than the send time

    dshot timeouts should be no longer than the pulse interval
    use correct timestamp for dmar send time

DavideLentini, 2022-12-07:
@andyp1per 
Thanks , I tried restoring the file before your commit in branch Copter-4.3.1 , it still seems to present the same problem even with the restored file . With a led_count higher than 44 , the drone startup process crashes and we can no longer connect via usb or via udp .
So I believe that the commit between 4.3.1 and 4.2.3 , which creates the problem I am experiencing , changed other parts of the code

DavideLentini, 2022-12-07:
@andyp1per 
What I would like to do is to use the git bisect tool to identify the commit that creates this problem , the problem is that my good commit is located in another branch (Copter-4.2.3) , compared to where my bad commit is located (Copter-4.3 branch).
So I find it difficult to exactly find the commit that creates the problem in this situation . I have tried going back in the master branch to the Copter-4.2.3 Release Notes commit and doing a checkout but the problem still persists

DavideLentini, 2022-12-07:
@andyp1per @ntamas 
Hello ! 
Then maybe we found something .
Came the doubt that the problem was between some parameter , we tried to change the parameter SCHED_LOOP_RATE from 800 to 400 . And the problem seems to solve , the drone starts normally and I can connect via usb . Also with the value put at 400 all combinations with the SERVO_DSHOT_RATE parameter work , on the contrary with the parameter SCHED_LOOP_RATE at 800 no combination with the SERVO_DSHOT_RATE works .

However, this is a completely different situation from Copter-4.2.3 , where we had kept the loop at 800 without ever having had any problems . 
I am however creating a topic on ardupilot discuss so that other developers can also investigate this problem .

rmackay9, 2022-12-07:
@DavideLentini,

We've got some performance issues in Copter-4.3.0 ([discussion1](https://discuss.ardupilot.org/t/internal-errors-0x100000-l-917-flow-of-ctr/87033), [discussion2](https://discuss.ardupilot.org/t/internal-errors-0x100000-i-921-flow-of-ctr-twitching-motion-in-loiter-during-forward-flight/93126)) that are our top priority to fix and we should be publishing some test binaries in the coming days on one of those two discussions.  Maybe you could try those to see if they help?

DavideLentini, 2022-12-08:
@rmackay9 
Yes absolutely , we are interested in these alpha test binaries . I confirm you , as I had also written above , that we basically have the same problem , with the main loop at 800 the drone can not start , putting it at 400 unlocks . 
I will send you the updated hwdefs for our luminousbee5 + a defaults.param file to compile a firmware bin as close as possible to what we usually compile .
I will write to you later today in private on discord. Thank you very much

andyp1per, 2022-12-08:
> of a full pulse for 100 LEDs will be around 3 msec only, right

At 400Hz the maximum slot you will be allowed is 2.5ms at 800Hz maximum allowed is 1.25ms. Default dshot rate is 1Khz so maximum allowed is 1ms. Dshot is constrained to be minimum 800Hz, so you only ever get 1.25ms maximum - but you never get that because dshot pulses get prioritized over LEDs - so the most you can ever hope for is 1.25ms - dshot pulse time.

andyp1per, 2022-12-08:
The solution here is to have a thread-per-timer - then the LEDs can run for as long as you want without disrupting dshot

DavideLentini, 2022-12-09:
@andyp1per 
Hello , thank you for the help you are giving us .

We understand that it is a problem that needs a long revision behind to be solved and that you have little time to devote to it .
So I would like to ask you the main points to be addressed in the ardupilot code to try to complete this project .

I offer to try to solve it , but you understand very well that first I need some tips , on the general plan to solve the problem and that I cannot go so blindly without knowing first where to put my hands in the code .

I will write to you on discord privately , so as not to clog up this issue , and I will post updates on my path here . 

Thank you very much