ArduPilot/ardupilot #22277: Error building copter with waf

suitangxiaoyi, 2022-11-24:
## Bug report

**./waf error **
When I build Sub-4.0, It reported some bugs to me

**Version**
Sub-4.0
git version 2.25.1
Ubuntu 20.04.5 LTS

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Logs**
kingpong@tangyi:~/ROV/400/ardupilot$ ./waf sub
Waf: Entering directory `/home/kingpong/ROV/400/ardupilot/build/Pixhawk1'
Padded 16 bytes for bootloader.bin to 16064
b'Embedding file sandbox.lua:libraries/AP_Scripting/scripts/sandbox.lua'
b'Embedding file io_firmware.bin:Tools/IO_Firmware/iofirmware_lowpolh.bin'
b'Embedding file bootloader.bin:/home/kingpong/ROV/400/ardupilot/Tools/bootloaders/Pixhawk1_bl.bin'
Checking for env.py
env added HAL_WITH_UAVCAN=1
env added PROCESS_STACK=0x2000
env added MAIN_STACK=0x400
env added IOMCU_FW=0
env added PERIPH_FW=0
env added BOARD_FLASH_SIZE=2048
env appended CPU_FLAGS=['-mcpu=cortex-m4', '-mfpu=fpv4-sp-d16', '-mfloat-abi=hard', '-u_printf_float']
env added APJ_BOARD_ID=9
env added APJ_BOARD_TYPE=STM32F427xx
env added FLASH_RESERVE_START_KB=16
env added CHIBIOS_BUILD_FLAGS=USE_FATFS=yes CHIBIOS_STARTUP_MK=os/common/startup/ARMCMx/compilers/GCC/mk/startup_stm32f4xx.mk CHIBIOS_PLATFORM_MK=os/hal/ports/STM32/STM32F4xx/platform.mk MCU=cortex-m4 ENV_UDEFS=-DCHPRINTF_USE_FLOAT=1
Waf: Leaving directory `/home/kingpong/ROV/400/ardupilot/build/Pixhawk1'
Build failed
Traceback (most recent call last):
  File "/home/kingpong/ROV/400/ardupilot/modules/waf/waflib/Runner.py", line 395, in task_status
    return tsk.runnable_status()
  File "/home/kingpong/ROV/400/ardupilot/Tools/ardupilotwaf/git_submodule.py", line 75, in runnable_status
    out = self.generator.bld.cmd_and_log(cmd, quiet=Context.BOTH, cwd=self.cwd)
  File "/home/kingpong/ROV/400/ardupilot/modules/waf/waflib/Context.py", line 461, in cmd_and_log
    raise e
waflib.Errors.WafError: Command ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/kingpong/ROV/400/ardupilot/modules/mavlink') returned 128

amilcarlucas, 2022-11-24:
did you do a
git submodule update --init --recursive
?

suitangxiaoyi, 2022-11-25:
I run it, but it seems doesn't work

 kingpong@tangyi:~/ROV/400/ardupilot$ git submodule update --init --recursive
kingpong@tangyi:~/ROV/400/ardupilot$

suitangxiaoyi, 2022-11-25:
`kingpong@tangyi:~/ROV/400/ardupilot$ git submodule update --init --recursive
kingpong@tangyi:~/ROV/400/ardupilot$ 
`

suitangxiaoyi, 2022-11-25:
after run 
git submodule update --init --recursive
the error changed
`[100/103] ChibiOS: Compiling bouncebuffer.c
[101/103] ChibiOS: Compiling watchdog.c
[102/103] ChibiOS: Compiling ch.cpp
[103/103] ChibiOS: Compiling syscalls_cpp.cpp

ChibiOS: Done!

../../modules/ChibiOS/ext/fatfs/src/ff.c: In function 'get_fat':
../../modules/ChibiOS/ext/fatfs/src/ff.c:1017:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1017 |    if (obj->objsize != 0) {
      |       ^
../../modules/ChibiOS/ext/fatfs/src/ff.c:1041:3: note: here
 1041 |   default:
      |   ^~~~~~~
arm-none-eabi-ar: creating modules/ChibiOS/libch.a

[  9/567] Compiling libraries/AC_AttitudeControl/AC_AttitudeControl_Sub.cpp
[ 10/567] Compiling libraries/AC_AttitudeControl/AC_AttitudeControl_Heli.cpp
[ 11/567] Compiling libraries/AC_AttitudeControl/ControlMonitor.cpp
[ 12/567] Compiling libraries/AC_AttitudeControl/AC_AttitudeControl_Multi.cpp
[ 13/567] Compiling libraries/AC_AttitudeControl/AC_PosControl_Sub.cpp
[ 14/567] Compiling libraries/AC_Avoidance/AP_OADatabase.cpp
[ 15/567] Compiling libraries/AC_Avoidance/AP_OAPathPlanner.cpp
[ 16/567] Compiling libraries/AC_Avoidance/AP_OABendyRuler.cpp
[ 17/567] Compiling libraries/AC_Avoidance/AP_OADijkstra.cpp
[ 18/567] Compiling libraries/AC_Avoidance/AP_OAVisGraph.cpp
[ 19/567] Compiling libraries/AC_Fence/AC_PolyFence_loader.cpp
[ 20/567] Compiling libraries/AC_PID/AC_PID_2D.cpp
[ 21/567] Compiling libraries/AC_PID/AC_PI_2D.cpp
[ 22/567] Compiling libraries/AC_PID/AC_HELI_PID.cpp
[ 23/567] Compiling libraries/AC_PID/AC_P.cpp
[ 24/567] Compiling libraries/AC_PID/AC_PID.cpp
[ 25/567] Compiling libraries/AC_Sprayer/AC_Sprayer.cpp
[ 26/567] Compiling libraries/AC_WPNav/AC_WPNav.cpp
[ 27/567] Compiling libraries/AC_WPNav/AC_Circle.cpp
[ 28/567] Compiling libraries/AC_WPNav/AC_WPNav_OA.cpp
In file included from ../../libraries/AP_Common/missing/cmath:3,
                 from ../../libraries/AP_HAL/CAN.h:23,
                 from ../../libraries/AP_HAL/HAL.h:17,
                 from ../../libraries/AP_HAL/AP_HAL_Main.h:19,
                 from ../../libraries/AP_HAL/AP_HAL.h:8,
                 from ../../libraries/AC_Avoidance/AP_OAVisGraph.h:5,
                 from ../../libraries/AC_Avoidance/AP_OAVisGraph.cpp:16:
/opt/gcc-arm-none-eabi-10-2020-q4-major/arm-none-eabi/include/c++/10.2.1/cmath:83:11: error: 'acos' has not been declared in '::'
   83 |   using ::acos;
      |           ^~~~
compilation terminated due to -Wfatal-errors.

In file included from ../../libraries/AP_Common/missing/cmath:3,
                 from ../../libraries/AP_HAL/CAN.h:23,
                 from ../../libraries/AP_HAL/HAL.h:17,
                 from ../../libraries/AP_HAL/AP_HAL_Main.h:19,
                 from ../../libraries/AP_HAL/AP_HAL.h:8,
                 from ../../libraries/AC_Avoidance/AP_OADatabase.h:3,
                 from ../../libraries/AC_Avoidance/AP_OADatabase.cpp:14:
/opt/gcc-arm-none-eabi-10-2020-q4-major/arm-none-eabi/include/c++/10.2.1/cmath:83:11: error: 'acos' has not been declared in '::'
   83 |   using ::acos;
      |           ^~~~
compilation terminated due to -Wfatal-errors.

In file included from ../../libraries/AP_Common/missing/cmath:3,
                 from ../../libraries/AP_HAL/CAN.h:23,
                 from ../../libraries/AP_HAL/HAL.h:17,
                 from ../../libraries/AP_HAL/AP_HAL_Main.h:19,
                 from ../../libraries/AP_HAL/AP_HAL.h:8,
                 from ../../libraries/AC_WPNav/AC_Circle.cpp:1:
/opt/gcc-arm-none-eabi-10-2020-q4-major/arm-none-eabi/include/c++/10.2.1/cmath:83:11: error: 'acos' has not been declared in '::'
[... 46 lines cut ...]
                 from ../../libraries/AP_Filesystem/AP_Filesystem_FATFS.h:7,
                 from ../../libraries/AP_Filesystem/AP_Filesystem.h:33,
                 from ../../libraries/AP_Terrain/AP_Terrain.h:20,
                 from ../../libraries/AC_WPNav/AC_WPNav.h:10,
                 from ../../libraries/AC_WPNav/AC_WPNav_OA.h:3,
                 from ../../libraries/AC_WPNav/AC_WPNav_OA.cpp:1:
/opt/gcc-arm-none-eabi-10-2020-q4-major/arm-none-eabi/include/sys/_default_fcntl.h:63: note: this is the location of the previous definition
   63 | #define O_CLOEXEC _FNOINHERIT
      | 

Waf: Leaving directory `/home/kingpong/ROV/400/ardupilot/build/Pixhawk1'
Build failed
 -> task in 'objs/AC_Avoidance' failed (exit status 1): 
	{task 140113847274624: cxx AP_OAVisGraph.cpp -> AP_OAVisGraph.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/AC_Avoidance' failed (exit status 1): 
	{task 140113847274176: cxx AP_OADatabase.cpp -> AP_OADatabase.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/AC_WPNav' failed (exit status 1): 
	{task 140113847366032: cxx AC_Circle.cpp -> AC_Circle.cpp.0.o}
 (run with -v to display more information)
 -> task in 'objs/AC_WPNav' failed (exit status 1): 
	{task 140113847366144: cxx AC_WPNav.cpp -> AC_WPNav.cpp.0.o}
 (run with -v to display more information)
`

peterbarker, 2022-11-27:
@suitangxiaoyi why are you trying to build a release which is 2,5 years old?

I'm not surprised that modern compilers can't cope with the older source code.

If you do want to compile this older code, try an older compiler; we used `gcc-arm-none-eabi-6-2017-q2-update` for a long time if memory serves.

I'd suggest moving to a more recent Sub release, however - and if you are doing development work then perhaps work from master.

suitangxiaoyi, 2022-12-01:
Thanks， I will try the master version

Williangalvani, 2024-02-23:
closing this as it is stale. feel free to open it again if the issue arises again

closed 2024-02-23 by Williangalvani