ArduPilot/ardupilot #22309: Internal Error - invalid argument in Scurves

bnsgeyer, 2022-11-28:
## Bug report

**Issue details**
Error occurred while executing landing during auto mission. Error was internal error 0x20000000 l:868 invalid argument. 
Aircraft completed autonomous landing without incident. Traced internal error to scurve.cpp

**Version**
Arducopter 4.2.3

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Traditional helicopter. 

**Hardware type**
cube orange

**Logs**
https://drive.google.com/file/d/13n4HSKRjBp28ASjPwfZTrLEopzUk6EW9/view?usp=drivesdk

rmackay9, 2022-12-13:
@bnsgeyer,

Does this happen with 4.3 and/or master?  I think we'd probably only investigate and try to resolve if it is happening in a later release.

bnsgeyer, 2022-12-13:
@rmackay9 Tried using his waypoint plan extracted from the log file in SITL with master and I couldn't reproduce the error.  I will look at it a little more tomorrow

lthall, 2022-12-13:
This error is called from line 868 in SCurve.cpp. This happens when a waypoint with effectively zero length is called. I think we have added protections to remove this problem. This may still happen if extremely short waypoints are called resulting in floating point accuracy problems. This error catches these problems and handles them without incident.

peterbarker, 2024-05-07:
I certainly haven't seen this problem for a long time, and @lthall thinks we fixed it.  OK to close, @bnsgeyer ?

rmackay9, 2024-05-08:
Thanks @peterbarker I'm going to go ahead and close this and we can re-open if necessary.  I haven't seen this reported in the forums at all.

closed 2024-05-08 by rmackay9