ArduPilot/ardupilot #22316: Performance reporting especially for F4 boards

rmackay9, 2022-11-29:
AP 4.3 introduced various improvements to features that come with an increased CPU cost compared to earlier versions.  This has led to some users (especially on lower end F4 CPUs including the CubeBlack) to start experience poor attitude and position control.

Below are some actions we think should help

Code changes
----------------

- [ ] High processor load warning message if over threshold
    - threshold based on difference between average loop rate vs actual loop rate?
    - text warning every 15 seconds?
    - enhance system status message to report "unhealthy" CPU?
- [ ] PM logging message should show variability in DT
- [ ] modify load calculation (which is both logged and reported in real-time to GCS?) so that it takes account of automatic loop slowdown.  E.g. if the main loop rate has been automatically slowed down from the user configured main loop rate the load should appear as very high
- [ ] add pre-arm if the main loop is running slowly
- [ ] write autotests that are run on real hardware
- [ ] add CI autotest using calgrind to improve tracking of CPU impact of a PR?  e.g. report before-vs-after count of instructions

Wiki changes
--------------

- [ ] recommend that users chose an autopilot with at least an F7 CPU?

References:

- Internal Error 917 is the "canary in the coalmine" warning that affected users see.  This is because the position controller raises this error if it is not updated for at least 3 (or 5?) main loop iterations.  E.g. a long main loop can trigger this internal error.

MichelleRos, 2022-11-29:
Perhaps a failsafe (eg landing the copter) rather than a pre-arm (or both) since the processor load can increase a lot in-flight?

andyp1per, 2022-11-29:
> Perhaps a failsafe (eg landing the copter) rather than a pre-arm (or both) since the processor load can increase a lot in-flight?

Right - you only see true load once armed

andyp1per, 2022-11-29:
Note that calgrind is of limited usefulness since the execution path in SITL/Linux is quite different to ChibiOS. Once thing we could do is build with thread statistics (--enable-stats) on both 4.2 and 4.3 and compare the the CPU difference (using ftp @SYS/threads.txt and @SYS/tasks.txt) to see if anything stands out. Also note I have done *a lot* of this for the QSPI setup, if anything 4.3 should be *better* than 4.2, so I am a bit puzzled as to why we are seeing this now. Maybe something was incorporated late cycle that broke things.

andyp1per, 2022-11-29:
Also one of my changes did not make it - https://github.com/ArduPilot/ardupilot/pull/19976

lthall, 2022-11-29:
This is what I have been telling people to do to try to get their cpu load back down to reasonable levels.

EK3_IMU_MASK 3 or 5 based on redundacy considerations (2 IMU’s)
INS_ENABLE_MASK 3 or 5 (2 IMU’s)
INS_USE 1 (not sure if they do anything any more)
INS_USE2 0 or 1 (not sure if they do anything any more)
INS_USE3 0 or 1 (not sure if they do anything any more)
INS_LOG_BAT_MASK 0
INS_FAST_SAMPLE 0
Limit harmonic notch’s to two harmonics
Don’t use any INS_HNTCH_OPTS
FFT_ENABLE 0
INS_GYRO_RATE 0
SCHED_LOOP_RATE 400

prathamesh7255, 2023-01-22:
@rmackay9 is this issue still open or the problem is solved by some other merge?