ArduPilot/ardupilot #22335: Default parameter values outside recommended range

IamPete1, 2022-12-02:
I recently came across some of the plane `PTCH_RATE_*` gains who's default values are outside the given range.

We now have the ability to get default values via FTP. I suggest we add param meta-data check that if a range is provided it much include the default value, I suspect we will find quite a few where this is not the case. Of course there may be some enable type parameters that default to a magic value, maybe we need some new markup for such ranges allowing them to skip this check. 

The main hurdle to this is the implementation of the param FTP protocol in autotest (or just going via mavproxy). We could also take the opportunity to add a check that both parameter methods return the same values.

007harshmahajan, 2022-12-08:
I would like to solve this issue, can you assign this to me ?

peterbarker, 2022-12-08:
```
[8:43 AM]Peter Barker: @Harsh_mahajan we can't assign it to you from a technical perspective.  But if you feel like working on it, it looks like it would be quite a useful autotest.  Generally I think you'll need to:
 - create a Python mavftp library out of MAVProxy's support for it
  - my branch here goes some way towards creating that library https://github.com/ArduPilot/MAVProxy/compare/master...peterbarker:MAVProxy:pr/fleet - let me know if you would like me to tody that up for you
 - write an autotest which includes that library from the MAVProxy libraries (we already include MAVProxy's terrain support in autotest in the same way) to fetch parameters and their defaults via mavftp
 - decode the resultant binary data (MAVProxy has facility to do that, presumably we'll need another library broken out for that!)
 - check the parameter metadata (which autotest has access to) against the downloaded parameter data to find inconsistencies such as the one @Peter Hall noticed
```