ArduPilot/ardupilot #22365: Arduplane Y4 configuration

tonysmartantzoulatos, 2022-12-08:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Tried testing Y4 configuration on arduplane for a vtol. It requires Frame type Tricopter and frame class Y4 (aka19). running multiple tests i found the problem listed below.

**Is your feature request related to a problem? Please describe.**
_Mission planner does allow tricopter on arduplane but does not have option 19 on the list. it also allows you to include option 19 and does not recognise it as an error. If set to option 19, then Motor 3 is not working. I verified this finding multiple times by changing my config on something that is on the list and uses 4 motors in which case all motors had pwm signals.

**Describe the solution you'd like**
_Include Y4 on arduplane.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

IamPete1, 2022-12-08:
Y4 is a quad frame class not tricopter

closed 2022-12-08 by IamPete1

tonysmartantzoulatos, 2022-12-08:
While checking the motor order on the wiki https://ardupilot.org/copter/docs/connect-escs-and-motors.html the Y4A is not listed on the Quad section, but right above Y6, which for Y6 you use Tri and not Hexa, but with Type 10 or 11 based on your config. If for sure it must be set as quad and not Tri, then probably this is what had me off, but i tested it as 1/19 and when i did i had zero yaw control (amd yes i checked multiple times that PWM signals were plugged into the right output)

tonysmartantzoulatos, 2022-12-08:
In addition to my previous comment i forgot to add this link that it was the only info i had found on this config https://discuss.ardupilot.org/t/quadcopter-in-tricopter-formation/87121/11 and shows 7/19 and not 1/19