ArduPilot/ardupilot #22371: want horizon lines on OSD

lotsofone, 2022-12-09:
## Feature request

**Describe the solution you'd like**
I want horizon lines to be implemented on OSD.

**Platform**
Plane

**Additional context**
Just like this
[image]

a1axx, 2022-12-10:
You would need a pixel / graphical type osd.

pitlab osd will give you this and work nicely with ardu.

lotsofone, 2022-12-11:
> You would need a pixel / graphical type osd.
> 
> pitlab osd will give you this and work nicely with ardu.

Does it require some hardware? From what I can understand now, is it something like CPU vs GPU? Such horizon lines require graphics ability from GPU so it cannot be implemented in ardupilot itself?

a1axx, 2022-12-11:
> GPU

Ardupilot standard FCs have no osd hardware,

Matek and others have the Max chip, character osd (cannot do graphical elements)

Pitlab definitely does graphical osd with full customisation. This is the additional hardware I use with an matek f765 wing. Even though the matek has osd, I have to bypass it and use the pitlab for osd. Full info here: 

https://www.rcgroups.com/forums/showthread.php?1811191-Pitlab-FPV-System-OSD-ILS-Autopilot-BuddyFlight-Ground-Station-Diversity-Ant-Tracker/page150

https://www.pitlab.com/pitlabshop/fpv-system/osd-mavlink-kit-detail.html

Using an Frsky pixel osd may be possible, however I have never tried this with ardupilot.

closed 2022-12-13 by lotsofone