ArduPilot/ardupilot #22374: DJI O3 MSP DP Osd character set support

a1axx, 2022-12-10:
Recently discussed and closed here:
https://github.com/ArduPilot/ardupilot/issues/22362

While the OSD will display in the goggles 2 with O3, the characters have some mismatches/errors.

Specific support for O3 osd may be required.
Character errors:

[image]

[image]

Supported characters:

[image]

Incompatible elements may need to be removed or mapped to the correct characters for correct display.

Thanks

a1axx, 2022-12-15:
Some work has been done here on Inav

https://github.com/iNavFlight/inav/pull/8631#issuecomment-1348653058

hyperion11, 2023-06-04:
agree. Incompatible elements are annoying

mf0o, 2023-12-28:
Hi @a1axx.

Where/How did you get the table for the available glyphs from? 
I'm currently working on some improvements on the icons (DJI Goggles2 + O3) but I fear some of those glyphs are not available. Maybe a firmware update changed something or so.

I would like to re-create that table with G2 goggles however and would appreciate any assistance.

Thank you in advance.

schumixmd, 2024-08-11:
what about, at least, remapping the characters for the home arrow, which points backwards when using DJI O3?