ArduPilot/ardupilot #22379: IRC Tramp - can not control VTX power

Ivosch, 2022-12-11:
## Bug report
Cannot set maximum (1600mW) VTX power, it works somehow but not properly. 800mW and lower powers are ok when setting VTX_POWER. Controlling power with RC11 is somehow inconsistent. When RC11 is at max level (2000), OSD reports 1000mW. For RC11 at 1000 it is 0mW. Channel and band control works fine.

I have same VTX on betaflight, settings:
[image]

Communication seems to be set properly
[image]

I tried to ask in forum, https://discuss.ardupilot.org/t/matek-h743-irctramp-1600mw-doesnt-work/93157, but no reactions.

**Version**
ArduPlane V4.3.1 (7d0c6e4d)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Wing - [ZOHD talon gt](https://www.zohd.net/zohd-talon-gt-rebel)

**Hardware type**
[Mateksys H743-wing v3](http://www.mateksys.com/?portfolio=h743-wing-v2#tab-id-6)
VTX [Jhemcu RuiBet Tran-3016W 25-1600mW](https://www.banggood.com/JHEMCU-RuiBet-Tran-3016W-5_8Ghz-48CH-PIT-or-25MW-or-200-or-400-or-800-or-1600MW-FPV-Transmitter-Built-in-AGC-Microphone-Support-IRC-For-RC-Racing-Drone-p-1936777.html?gmcCountry=US&currency=USD&utm_source=googleshopping&utm_medium=cpc_eu&utm_content=dcr&utm_campaign=aceng-pmax-mixcountry-creative-page-nofeed-220816&ad_id=&gclid=Cj0KCQiAsdKbBhDHARIsANJ6-jeJ76SbdG1d1DJB_7wX8TTtm6fCjqRu8pMoclDNSAeOg7qv-gut-DcaAmkqEALw_wcB&cur_warehouse=CN)

**Logs**
What logs do you need? How to download it?

Ivosch, 2023-01-07:
anyone?

oloendithas, 2023-04-24:
The weird thing I noticed is that it totally breaks if I set RC*_OPTION=94. The best I was able to get out of it is to switch automatically between Pit and MAX power on arm/disarm.
My chaotic desperate settings are: Half-Duplex, TX_PullDown, TX_NoDMA, Pit, Pit on DIsarm.

Kopervlad, 2023-06-07:
Hello all,
I had an isuue just like Topic Starter (Ivosch).
I can to control my VTX from INAV. But I can't in ArduCopter (Mission Planner 1.3.8479.20539).

**Version**
ArduCopter V4.3.7 (c8506ed4)

**Platform**
[ ] All
[ ] AntennaTracker
[X] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

**Airframe type**
X-CopterFrame

**Hardware type**
FC [Matek F405-TE](http://www.mateksys.com/?portfolio=f405-te)
VTX [Jhemcu RuiBet Tran-3016W 25-1600mW](https://www.ebay.com/itm/275069614832) The same like TS has.

**Logs**
What logs do you need? How to download it?

I was trying to activate and setup VTX control using IRCTramp. It didn’t work at me too. Even BAND and Channel. And is it true that MP have no possibility to set POWER more than 1000 mW?

Configured parameters:
[image]
and with SERIAL5_OPTIONS 0 was tried also (and many other :) )
and
[image]

and RC14_OPTION set to 94

Help me please to solve this problem.
With love from Ukraine!

Kopervlad, 2023-06-07:
I don't know how to do it yet. Will learn.
So You mean that INAV FW has it built in and ArduCopter doesn't?

andyp1per, 2023-06-07:
ArduCopter on 2Mb boards has it built in, on 1Mb boards it does not

https://custom.ardupilot.org/

Kopervlad, 2023-06-07:
Thanks!
I did a custom FW with IRC. It works now.
But still can't increase Power more than 1000 mW.
Looking for solve.

Kopervlad, 2023-06-08:
Thank you very much for your help.
In my case it is very important to switch at least between 25/400/1600.
Let me remind you, I am from Ukraine))) and this is really one of the important tasks for me.
Maybe some logical switches etc.?

Kopervlad, 2023-06-08:
> ArduCopter on 2Mb boards has it built in, on 1Mb boards it does not
> 
> https://custom.ardupilot.org/

I will be very grateful for your help

pereverni, 2024-06-30:
> Thanks! I did a custom FW with IRC. It works now. But still can't increase Power more than 1000 mW. In my case 800 mW respectivly (nearest to low side). VTX power levels PIT/25MW/200/400/800/1600MW Looking for solution.

Hey man, I need some help with the exact same question. Would be grateful for your help. I'm from Ukraine too btw

andyp1per, 2024-06-30:
Fixed by https://github.com/ArduPilot/ardupilot/pull/26923

closed 2024-06-30 by andyp1per

yuriystupak2020, 2024-12-25:
> The weird thing I noticed is that it totally breaks if I set RC*_OPTION=94. The best I was able to get out of it is to switch automatically between Pit and MAX power on arm/disarm. My chaotic desperate settings are: Half-Duplex, TX_PullDown, TX_NoDMA, Pit, Pit on DIsarm.

> > In my case it is very important to switch at least between 25/400/1600.
> 
> Ok, I've made a LUA script and tested it. Looks working. You gonna have to assign some 3-pos switch to scripting_300. On my setup the actual switch happens with a slight delay, in around 5 seconds, also looks sensitive to quick switching, may not switch to the last value. Also you can display the current VTX power on OSD. Again, you'll need to disable all PIT options from VTX_OPTIONS. ![image](https://private-user-images.githubusercontent.com/5385124/244431583-86540c76-bd27-4183-8601-6856e2fa6a53.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUxMzk0NjgsIm5iZiI6MTczNTEzOTE2OCwicGF0aCI6Ii81Mzg1MTI0LzI0NDQzMTU4My04NjU0MGM3Ni1iZDI3LTQxODMtODYwMS02ODU2ZTJmYTZhNTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjVUMTUwNjA4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTZiNjJmNjNlZmI4NjM3MjZkMTgwZmJkMGVjZjQwMjM2OGViZjE3OGRkMTI2ZDU0MTBjMDE4YTMzMDE1ZjliMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.S06o-GfAWNYWjxqAPkHX2G7dwDqS8srtino2t0gUVOU)
> 
> ` local scr_rc_300 = rc:find_channel_for_option(300) local VTX_POWER = Parameter() --creates a parameter object VTX_POWER:init('VTX_POWER') --get the physical location in parameters memory area so no future name search is done
> 
> function vtx_switch() local rc_connected = rc:get_pwm(2) > 0 and rc:has_valid_input() and scr_rc_300 if rc_connected then local sw_pos = scr_rc_300:get_aux_switch_pos() vtx_toggle(sw_pos) end end
> 
> function vtx_toggle(sw_pos) local current = VTX_POWER:get() --retrieve that parameters value and assign to "parameter" if sw_pos == 0 then if current ~= 25 then gcs:send_text(0, "Current VTX power " .. tostring(current)) gcs:send_text(0, "Setting VTX power to 25") end VTX_POWER:set(25) elseif sw_pos == 1 then if current ~= 400 then gcs:send_text(0, "Current VTX power " .. tostring(current)) gcs:send_text(0, "Setting VTX power to 400") end VTX_POWER:set(400) else if current ~= 1600 then gcs:send_text(0, "Current VTX power " .. tostring(current)) gcs:send_text(0, "Setting VTX power to 1600") end VTX_POWER:set(1600) end end
> 
> function update() vtx_switch()
> 
> return update, 1000 end
> 
> return update() `
> 
> How familiar you with LUA scripts? Do you need help with applying it? I'm not sure if F405 can handle scripting, but if it can, maybe the HEAP size should be lowered.

hello. could you help me too man?