ArduPilot/ardupilot #22383: AP_Arming: AHRS and GPS differ by xxx when Non-GPS is used

rmackay9, 2022-12-12:
If a vehicle is setup for non-GPS navigation and GPS_TYPE = 0 a pre-arm errors triggers saying something like, "AHRS and GPS differ by x".  The vehicle appears correctly on the map and should be flyable but the pre-arm warning prevents this forcing the user to set ARMING_CHECKS = 0.

I discovered this issue when using the [ModalAI VOXL setup as described here on the wiki](https://ardupilot.org/copter/docs/common-modalai-voxl.html).

IamPete1, 2022-12-12:
Similar to https://github.com/ArduPilot/ardupilot/issues/14661

peterbarker, 2024-11-23:
Looks like this one (and the one @IamPete1 links to) might have been resolved with #22393 .

Closing this one now

closed 2024-11-23 by peterbarker