ArduPilot/ardupilot #22391: Flip - Flight Mode

CarlouT, 2022-12-13:
## Feature request
At the end of the flip, the copter will climb. Is it possible to control how much the copter rises during the climb (increase or decrease the climb distance depending on a parameter)? Is it also possible to add another parameter to say where the copter will flip (i.e. to the right, left, forward or backwards)?

**Describe the solution you'd like**
Add a parameter that can controls the climb rate of the flip. Currently according to @rmackay9, it is hard coded to 20%. Can this hardcoded value be assigned to a parameter? And add another parameter that will tell where the flip will rotate (i.e. to the right, left, forward or backwards).

**Describe alternatives you've considered**
There is no alternative because the settings is hard coded.

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
The flip flight mode as I have tested is perfect except that at the end of the flip, the copter rises a lot which is not so pretty. It would be nice if the rise can be controlled. This is currently defaulted to 20% https://discuss.ardupilot.org/t/flip-flight-mode/94132/2?u=carlou. The current flight mode also is defaulted to flip to the right not unless you can move the elevator or aileron stick to the side you want to flip. Can this attitude also be assigned to a parameter?

Nickmacd88, 2022-12-22:
I will work on this request.

My current plan is to create a parameter from -1.0-1.0 which can adjust the throttle up or down from the pilot's initial throttle input at the end of the flip. I will default this value to 0, which will exit the flip with the same throttle setting as the entry and should give the pilot the same altitude as prior to entry.

I will also create a parameter to specify direction, leaving it defaulted to the current flip direction.

Caveating this plan of action that this is my first attempt at making a contribution to ArduPilot.

CarlouT, 2022-12-22:
Looking forward for this feature soon.

Nickmacd88, 2022-12-26:
I have a mostly working solution for this problem.

Quick question, in the documentation the flip is defaulted to the left. However, in the implementation, if the roll input was at zero the copter would flip right.

Do you want me to leave the default direction to the left IAW the documentation or leave it to the right because that's what the end users are used to?

rmackay9, 2022-12-26:
@Nickmacd88,

Txs for noticing the documentation issue.  I think it is better to fix the docs than the code..

Nickmacd88, 2022-12-28:
https://github.com/ArduPilot/ardupilot/pull/22515

Pull request has been created.