ArduPilot/ardupilot #22392: Copter: terrain offset clamped when using Non-GPS navigation

rmackay9, 2022-12-13:
When testing the ModalAI VOXL camera I noticed that our terrain offset clamping triggers meaning that if a user attempts an autonomous mission with terrain altitudes the vehicle could fly incredibly high.

The underlying issue may be that Mission Planner (and perhaps other GCSs) do not provide an accurate altitude when they set the origin.  They may set the altitude as zero.

[image]

alfiianaa, 2023-03-20:
how about using GPS navigation ?

rmackay9, 2023-03-20:
It turns out the issue is in Mission Planner and I've raised an issue https://github.com/ArduPilot/MissionPlanner/issues/3083

When using GPS navigation the "clamping" message can still be dangerous in that the vehicle may fly at the wrong altitude if it is executing auto mode missions with altitude type set to Terrain.  We recently relaxed the limit to make this less likely and this change will be included in 4.4. https://github.com/ArduPilot/ardupilot/pull/23176

closed 2023-03-20 by rmackay9