ArduPilot/ardupilot #22405: RC Inputs & Outputs freeze (crashlog)

simon88, 2022-12-14:
## Bug report

**Issue details**
I suffered a RC In and Out freeze caused a crash on my hexa copter. Old issue with the same problem [https://github.com/ArduPilot/ardupilot/issues/5859](url)

[image]

**Version**
4.2.2

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Hexacopter

**Hardware type**
Orange Cube

**Logs**
[https://drive.google.com/file/d/11bzhfpx8oTY5SzQqDWa90Ej0akxuVKjA/view?usp=sharing](url)

IamPete1, 2022-12-14:
IOMCU failed, hard to say why.

simon88, 2022-12-15:
it's frustrating to crash an expensive drone and not know why...

IamPete1, 2022-12-15:
> Is it a coincidence that the bug always appears on hexa ?

Yes

There are some more steps we could take to debug, the subsequent log would tell us if there was a watchdog reboot. You could test if the IOMCU works now, or if it is permanently broken.