ArduPilot/ardupilot #22482: ardupilot_ros can not start correcttly

lukezhqin, 2022-12-23:
## Bug report

**Issue details**

_I follow https://github.com/ArduPilot/ardupilot_ros instructions to install all good environment but I have to make some changes to the scripts. Two questions below:_

_this line   [     rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y](https://github.com/ArduPilot/ardupilot/blob/22f45675693991bf0021d5e55697dc611770c27a/Tools/environment_install/install-ROS-ubuntu.sh#L238), I have to add something behind to finish this installation. like: rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y -i -r, which is refered to https://answers.ros.org/question/194139/turtlebot-2-indigo-installation-problem-cannot-locate-rosdep-for-turtlebot_stdr-turtlebot_dashboard-and-turtlebot_gazebo/_

_When I finally finished all installation. I run first code like "roslaunch ardupilot_ros gzbo.launch"_

_Then it stopped with messages below:_
_checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:40497/

SUMMARY

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: noetic
 * /rosversion: 1.15.15
 * /use_sim_time: True

NODES
  /
    base_to_laser_broadcaster (tf/static_transform_publisher)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [14361]
ROS_MASTER_URI=http://localhost:11311/

setting /run_id to 73dbd6cc-8264-11ed-a72e-99c05ee5c545
process[rosout-1]: started with pid [14371]
started core service [/rosout]
process[gazebo-2]: started with pid [14374]
process[gazebo_gui-3]: started with pid [14379]
process[base_to_laser_broadcaster-4]: started with pid [14387]
[ INFO] [1671760348.002861324]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1671760348.003973909]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1671760348.112888983]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1671760348.114152304]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting..._

**Version**
_Copter-4.3.1_

**Platform**
[×] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_quad_

**Hardware type**
_ros sitl_

**Logs**
_Please provide a link to any relevant logs that show the issue_

lukezhqin, 2022-12-23:
@rmackay9 can you kindly help me check this bug? Thanks a lot!

Ryanf55, 2024-11-21:
If this needs fixed, it must be through `vcstool` for the ardupilot team to support it. 

ROS 1 does not distribute the package on any ROS 1 disto I tested
```
(venv-ardupilot) ryan@B650-970:~/Dev/ardu_ws/src/ardupilot$ rosdep resolve --rosdistro=melodic cartographer_ros
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: no rosdep rule for 'cartographer_ros'
(venv-ardupilot) ryan@B650-970:~/Dev/ardu_ws/src/ardupilot$ rosdep resolve --rosdistro=noetic cartographer_ros
ERROR: no rosdep rule for 'cartographer_ros'
(venv-ardupilot) ryan@B650-970:~/Dev/ardu_ws/src/ardupilot$ rosdep resolve --rosdistro=humble cartographer_ros
#apt
ros-humble-cartographer-ros
```

 I recommend using ROS 2 for now because it's available there. We also have a nice wiki for it.
 https://ardupilot.org/dev/docs/ros2-cartographer-slam.html
 
 It's built in CI too, so we know it works.

We do not have any dev team members using ROS 1 to support this. 

Closing as "won't fix" so we can clean up our issue backlog.

closed 2024-11-21 by Ryanf55