ArduPilot/ardupilot #22486: Possible error in quaternion update formula.

slovak194, 2022-12-23:
Dear @tridge,

I'm following the derivation of the EKF3 and found this, possible, error in how quaternion is updated with d_ang_true.

https://github.com/ArduPilot/ardupilot/blob/0816d5306db0847fdbee8b04bea4a4dac2faf6a9/libraries/AP_NavEKF3/derivation/main.py#L634

I assume the formula **dq = (1/2) * q * w** is used here, and **w** has to be expressed as quaternion [0, w[0], w[1], w[2]] (scalar part = 0). But that is not the case.
In my opinion this line should look like:
`q_new = quat_mult(q, Matrix([0, 0.5 * d_ang_true[0],  0.5 * d_ang_true[1],  0.5 * d_ang_true[2]]))`
Please, correct me if I'm wrong. I might miss something.

Best regards,
Alex

tridge, 2023-01-08:
@priseborough can you take a look?

priseborough, 2023-01-08:
We are converting the delta angle vector to a quaternion and then using the quaternion product rule to propagate the quaternion states to the current time step. The scaler part is 1 because the quaternion representing the small angle rotation from previous to current time step must have a norm of 1. We use the small angle approximation for the delta quaternion in the filter derivation for computational efficiency. Note that the INS state prediction in the EKF uses the exact form.

closed 2023-01-08 by priseborough