ArduPilot/ardupilot #22517: Possibility to use a RangeFinder facing up for copter altitude

simon88, 2022-12-28:
## Feature request

**Is your feature request related to a problem? Please describe.**
During indoor flight sometimes it would be more efficient to use the ceiling rather than the floor. Whether for lidar or optflow.

**Describe the solution you'd like**
Possibility to use a rangefinder oriented facing up for indoor flight use to control altitude of copter,  like it's already done If a rangefinder is oriented facing down.

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

amilcarlucas, 2022-12-28:
This is already supported. google for "arducopter surface tracking"

closed 2022-12-28 by amilcarlucas

simon88, 2022-12-28:
Yes @amilcarlucas but if I look here [https://ardupilot.org/copter/docs/terrain-following-manual-modes.html](url) in setup configuration we must have Connect a downward facing lidar.
 I can't see anywhere the possibility to do the same with a upward lidar.

rmackay9, 2022-12-28:
I've added a wiki issue to add docs on how to setup upward facing surface tracking https://github.com/ArduPilot/ardupilot_wiki/issues/4841

simon88, 2022-12-29:
Perfect thanks @rmackay9