ArduPilot/ardupilot #22522: Build: Can not build with python 3.11 because of outdated waf version

Szymon-SR, 2022-12-29:
## Bug report

**Issue details**
Issue occurs when trying to build ardupilot for SITL usage, on Fedora 37 Linux, with system python version 3.11.1
On fresh ardupilot clone, waf version is 2.0.9. When running ``` ./waf plane ``` an error occurs:
```
File "/home/szymon/ardupilot/modules/waf/waflib/Utils.py", line 231, in readf
    with open(fname, m) as f:
         ^^^^^^^^^^^^^^
ValueError: invalid mode: 'rUb'
```
U mode is no longer valid in python 3.11.

As workarounds, I have tried running with python 3.9 instead of 3.11, and manually changing the Utils.py file to 'rb', which both solved the issue, however the issue with deprecated file read mode appears to be fixed in newer versions of waf:
https://gitlab.com/ita1024/waf/-/commit/68997828c850ce7fb30b73b4adfde35053e539d1

Running with waf version 2.0.24 also solved the issue for me. Currently cloning ardupilot provides an older waf version: 2.0.9.

**Version**
Current source code - master branch

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
None - SITL

**Hardware type**
SITL

**Logs**

cj-chua, 2023-04-28:
@Szymon-SR How did you update waf to 2.0.24?

Ryanf55, 2023-04-28:
Another user on Mac is also facing this: 
`Mac M1 Ventura (but using a x86 terminal), I’m building master`

cj-chua, 2023-04-28:
Can confirm that the python version was indeed the root cause. I was using python `3.11` on Mac M1 x86 terminal with no success. 

I fixed by installing pyenv, using python `3.10.4` as specified in the install script, and then modify the pip command in the script to use `pyenv exec pip` instead of just `pip`. Doing so i got it to build without having to modify waf

tridge, 2023-05-01:
fixed by this PR: https://github.com/ArduPilot/waf/pull/12

closed 2023-05-01 by tridge

derpeter, 2023-09-18:
This fix has not made it into the Copter-4.4.0 release. Doing the changes manual fixed it for me.
(Im not familar with the arudpilot development flow so not sure if that an mistake or intentional)