ArduPilot/ardupilot #22543: Notch Filtering seems broken

Hwurzburg, 2022-12-31:
while setting up a miniquad to get some wiki screenshots of using pre/post batch sampling for notch filter setup, found that the throttle notch did not seem to work....I tried a fixed notch also, same results...here are the FFTs using mavfft_isb.py:

Pre filter sampling:
[image]
Post filter sampling with no notch enabled:
[image]
Post filter with notch enabled and set to fixed at 178Hz:
[image]
Notch supposedly active at 178Hzfrom the logging of FTNS.NF:
[image]
Notch params:
MAV> INS_HNTC2_ENABLE 0.000000 # Disabled
INS_HNTCH_ATT    40.000000
INS_HNTCH_BW     90.000000
INS_HNTCH_ENABLE 1.000000 # Enabled
INS_HNTCH_FM_RAT 1.000000
INS_HNTCH_FREQ   178.000000
INS_HNTCH_HMNCS  3.000000 # 1st harmonic|2nd harmonic
INS_HNTCH_MODE   0.000000
INS_HNTCH_OPTS   0.000000 # 
INS_HNTCH_REF    1.000000

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
this is on master as of 12/30/2022

andyp1per, 2022-12-31:
@Hwurzburg please retry with a much higher LPF - 20Hz is not really going to allow you to see what the notch is doing. Please also do a long hover to see the effect (e.g. 2mins). We have tests that indicate that the notch is working, so will need a bit more proof than this.

Hwurzburg, 2022-12-31:
with higher LPF and longer hover, I can see the gyro notch now....need to change the wiki a bit in light of this and make sure that increasing the LPF cutoff is needed to help see the results and is the point of the notches....thanks

closed 2022-12-31 by Hwurzburg