ArduPilot/ardupilot #22549: FS_EKF_ACTION: EKF Failsafe Action additional sensors

Tangoviktor, 2023-01-01:
## Feature request

The normal EKF failsafe action is either land or althold. Would it be possible to add flow-hold as an 4th option?
It would only work on the presence of an flow meter. An alternative solution for it would be to add flow sensor information to the althold flight mode, if the flow sensor is present.

rmackay9, 2023-01-02:
@Tangoviktor,

In general I'm trying to steer users away from using the flow hold flight mode because it does not work reliably on many vehicles.  I wonder if you've successfully used the flowhold flight mode?

Perhaps the intention is to allow fallback to optical flow if the GPS fails?  If this is the intention AP supports [GPS/Non-GPS Transitions using Lua scripts](https://ardupilot.org/copter/docs/common-non-gps-to-gps.html).

rmackay9, 2023-01-03:
@Tangoviktor,

OK good.  I think I will close this issue then although I will still see any comments made.

[The GPS/Non-GPS transitions page includes a link to a script for GPS <-> Optical flow transitions.](https://ardupilot.org/copter/docs/common-non-gps-to-gps.html) (search for "ahrs-source-gps-optflow.lua")

closed 2023-01-03 by rmackay9