ArduPilot/ardupilot #22555: BUG: RTL behaviour w/poly fence and OA

GlidLov, 2023-01-02:
## Bug report

hi all
i think i found a bug, or eithr i need full understanding of GeoFence function. Here i describe my use case:
i usually work with polygonal geofence to define a "L" shaped flight area (i use QgroundControl for that). 
If i turn ON fence, but turn OFF "stop at fence" (AVOID_BEHAVE), copter will break fence and switch to RTL, which is correct.

However if i set OA_TYPE different than 0, (which makes bendyruler or dijkstra active) copter will stop out of fence in RTL, behaving like it's looking for a path, indefinetely. If i turn OFF OA_TYPE ( =0) it will came back correctly. I think in this case the correct behaviour is to came back into fence (shortest path, orthogonal to fence line) and THEN look for a shorthest path to to RTL  (which can be sliding on the fence wall either, or other path) 
other than this, if the take off point and actual position in the "L" are on the far edges, and avoid is set to "SLIDE" , copter will not slide but will break the fence 
while i'm looking to make it slide on the edge of fence. I could not find a proper setting for this. Tried on SITL 4.4.0Dev

**Version**
SITL 4.4.0

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rishabsingh3003, 2023-01-08:
I am not sure I understand the problem fully. Is it that the current behavior of our path-planning libraries is wrong when RTL is activated from OUTSIDE the fence?

GlidLov, 2023-01-09:
Rishab, that's correct. This happens when veichle breach the fence and RTL is engaged, considering there's some inertia and delay before RTL, you came out of fence by some meters before RTL kicks in, then veichle seems to have "lost" its path