ArduPilot/ardupilot #22559: Build fails for H7 processors.

k5beason, 2023-01-03:
## Bug report

**Issue details**
Unable to build H7 builds. MatekH743-rover, CubeOrange-rover fail. I can build CubePurple-rover. M1 Mac on Ventura
 
**Version**
4.3
**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  x] Rover
[  ] Submarine

**Airframe type**
Rover
**Hardware type**
MatekH743
**Logs**
In file included from ../../libraries/AP_NavEKF2/AP_NavEKF2_core.h:30:0,
                 from ../../libraries/AP_NavEKF2/AP_NavEKF2.cpp:3:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'

[739/850] Compiling libraries/AP_NavEKF3/AP_NavEKF3_Outputs.cpp
In file included from ../../libraries/AP_NavEKF3/AP_NavEKF3_core.h:31:0,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3_VehicleStatus.cpp:3:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'

[740/850] Compiling libraries/AP_NavEKF3/AP_NavEKF3_GyroBias.cpp
In file included from ../../libraries/AP_NavEKF2/AP_NavEKF2_core.h:30:0,
                 from ../../libraries/AP_NavEKF2/AP_NavEKF2_OptFlowFusion.cpp:5:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'

[741/850] Compiling libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp
In file included from ../../libraries/AP_NavEKF3/AP_NavEKF3_core.h:31:0,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3_Outputs.cpp:4:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
In file included from ../../libraries/AP_Math/matrix3.h:40:0,
                 from ../../libraries/AP_Math/AP_Math.h:13,
                 from ../../libraries/AP_Common/Location.h:3,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3.h:22,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3_Outputs.cpp:3:
../../libraries/AP_NavEKF3/AP_NavEKF3_Outputs.cpp: In member function 'void NavEKF3_core::send_status_report(GCS_MAVLINK&) const':
../../libraries/AP_Math/ftype.h:24:24: error: narrowing conversion of 'fmax(fmax(((double)magVar.Vector3<float>::x), ((double)magVar.Vector3<float>::y)), ((double)magVar.Vector3<float>::z))' from 'double' to 'float' inside { } [-Werror=narrowing]
 #define fmaxF(x,y) fmax(x,y)
                        ^
../../libraries/AP_NavEKF3/AP_NavEKF3_Outputs.cpp:624:9: note: in expansion of macro 'fmaxF'
         fmaxF(fmaxF(magVar.x,magVar.y),magVar.z),
         ^
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

In file included from ../../libraries/AP_NavEKF3/AP_NavEKF3_core.h:31:0,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3_OptFlowFusion.cpp:4:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'

In file included from ../../libraries/AP_NavEKF3/AP_NavEKF3_core.h:31:0,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3_Control.cpp:4:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'

In file included from ../../libraries/AP_NavEKF3/AP_NavEKF3_core.h:31:0,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3.cpp:3:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'

In file included from ../../libraries/AP_NavEKF3/AP_NavEKF3_core.h:31:0,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3_MagFusion.cpp:4:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'

In file included from ../../libraries/AP_NavEKF3/AP_NavEKF3_core.h:31:0,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3_GyroBias.cpp:1:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'

In file included from ../../libraries/AP_NavEKF3/AP_NavEKF3_core.h:31:0,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp:4:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'

In file included from ../../libraries/AP_NavEKF3/AP_NavEKF3_core.h:31:0,
                 from ../../libraries/AP_NavEKF3/AP_NavEKF3_core.cpp:4:
../../libraries/AP_NavEKF/AP_NavEKF_core_common.h:64:30: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic error "-Wframe-larger-than=2500"
                              ^
../../libraries/AP_NavEKF3/AP_NavEKF3_core.cpp: In member function 'void NavEKF3_core::CovariancePrediction(Vector3F*)':
../../libraries/AP_NavEKF3/AP_NavEKF3_core.cpp:1806:1: warning: the frame size of 2112 bytes is larger than 1300 bytes [-Wframe-larger-than=]
 }
 ^
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'

Waf: Leaving directory `/Users/kbeason/ardupilot/build/MatekH743'
Build failed
 -> task in 'objs/AP_NavEKF3/Rover' failed (exit status 1): 
        {task 4365371872: cxx AP_NavEKF3_Outputs.cpp -> AP_NavEKF3_Outputs.cpp.7.o}
 (run with -v to display more information)

rmackay9, 2023-01-03:
I'm able to build CubeOrange for any platform using Windows 11.

k5beason, 2023-01-03:
I was successful in Ubuntu 22.04 VM. Issue seems specific to my M1 mac. I'll close close the issue.

closed 2023-01-03 by k5beason

rmackay9, 2023-01-03:
@k5beason,

Thanks for your persistence and feedback!

johnreine, 2023-02-03:
@k5beason ,
I am getting this same error on a Mac M2.  What did you do to fix it?

timtuxworth, 2023-02-03:
I have an existing build environment on my Mac M1Pro running Ventura, I just built rover for MatekH743 on master. I think maybe something is up with your setup.
[image]

k5beason, 2023-02-03:
I ended up just building in a Linux environment. Love the M series macs but they don’t always play nice with open source software with lots of dependencies.Sent from my iPhone

johnreine, 2023-02-03:
Thank you!  I agree - something is off.  Will dust off my linux machine.  (haven't gotten VM working on M2 yet)