ArduPilot/ardupilot #22564: GCS_MAVLINK send_autopilot_state_for_gimbal_device(): feed_forward_angular_velocity_z incorrect

olliw42, 2023-01-03:
To help gimbals with their stabilization and control, the autopilot can send the mavlink message `AUTOPILOT_STATE_FOR_GIMBAL_DEVICE` to the gimbal, which provides various information on the vehicle status, among them the field `feed_forward_angular_velocity_z`.

The purpose of this field is admittedly not very precisely explained in the message's description nor the documentation, but the intention is clear. It is supposed to tell the gimbal what the vehicle has been _**commanded**_ to yaw (this info can allow the gimbal to do various tasks better)(like e.g. pan/follow). Note that this is quite different from the _**actual**_ yawing of the vehicle, as determined e.g. by the vehicle's sensors. The latter info can in fact be already obtained from other data fields, specificially `angular_velocity_z` (or if one would want by differencing the quaternion)(or ATTITUDE)(or...).

Currently, this field is populated from `vehicle->get_rate_bf_targets(rate_bf_targets)` (https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L5362-L5367). This however does **_not_** properly reflect the intention to yaw, as commanded e.g. by a stick input or a mission command or alike, but rather reflects the actual yawing of the vehicle. It thus cannot serve the purpose at all, and in fact does not provide any benefit over just using the actual vehicle's yaw rate.

Given that `get_rate_bf_targets()` yields the setpoints for the rate controller, you may dispute this statement. I am not very familiar with the details of the ArduPilot pid controllers, and hence will not try to argue here (it though seems to me that the issue is the cascaded nature). Instead I will demonstrate this by data.

The data were produced with my fork of ArduCopter (BetaCopter), where I added logging messages to log the relevant data, and it is not using ArduPilot's mounts but my own STorM32 mount. However, I think this really doesn't change anything concerning the matter and all is true also for vanilla ArduCopter 
(for reference, the code: https://github.com/olliw42/BetaPilot/blob/BetaCopter43/libraries/AP_Mount/BP_Mount_STorM32_MAVLink.cpp)

[image]

The attached picture shows three lines:
- blue: yaw stick rc input
- green: feed_forward_angular_velocity_z. It is logged as MTL0.YawRate, but you will find it to be identical to RATE.YDes.
- red: the vehicle's actual yaw rate as measured by the gyro z axis.

The justification of the above statement is obvious: feed_forward_angular_velocity_z (aka RATE.YDes) clearly more closely reflects the vehicle's actual yaw rate than the yaw rate command. Q.E.D.

To close, I have to, and want to, admit that I drop this issue for entirely "selfish" reasons. I would want to use feed_forward_angular_velocity_z for better pan/follow modes in the STorM32 gimbal controller, but struggle since long to figure out how to get that info for that field in the code. So I was excited to see an implementation now, but am sad to see that it fails to serve the purpose. In short, I would be very interested in a solution, but myself lack the capability to figure it out. :)

lthall, 2023-01-04:
I think @olliw42 is correct here it should be _euler_rate_target.z but we don't have an accessor as we were not using it outside the Attitude controller.

I think the AUTOPILOT_STATE_FOR_GIMBAL_DEVICE is very short cited. If the target yaw rate is useful then so would roll and pitch. I mention this because the message is listed as a work in progress.

rmackay9, 2023-01-04:
Great, thanks for this feedback and advice.  I'll add it to my to-do list to fix.

lthall, 2023-01-05:
> Q1: while probably self-answering, just to have that 100% sure, this would be it for ALL possible ways of how to command yaw? I.e., sticks, missions, mavlinks, scripts, ...

If I understand your question correctly, yes this will work for all multirotor flight modes. However, this will not work for plane as it does not use AC_AttitudeControl.

Sorry I can't answer your second question.

> nobody was interested in this for pitch & roll

In these situation there are only two outcomes in the long run. Either yaw feed forward isn't as useful as people hope OR roll and pitch are also useful and should have been included in the message. It is normal for people to be short sighted when defining these messages and leave out data that they later discover is important.

olliw42, 2023-01-05:
> If I understand your question correctly, yes this will work for all multirotor flight modes. 

great. I was not asking only for the different flight modes, but also the different ways it is possible to command yaw, e.g. scripts, etc.. But I assume that by the nature of the code structure the answer will be yes.

> However, this will not work for plane as it does not use AC_AttitudeControl.

ah ... thx for clarifying this
that's sad :) any possible substitute?

> Sorry I can't answer your second question.

you did ... not used in plane, so needs to link to something else for plane :)

> > nobody was interested in this for pitch & roll
> 
> In these situation there are only two outcomes in the long run. Either yaw feed forward isn't as useful as people hope OR roll and pitch are also useful and should have been included in the message. 

I disagree. A third outcome in the long run can be that they were added based on some argument that there only can be these two outcomes but that they never ever will be used by any gimbal implementation. 

> It is normal for people to be short sighted when defining these messages and leave out data that they later discover is important.

well, thank you for this insight and it is great to see that you are not among the normal people, but I can tell you that those who were not interested are involved in implementing gimbals since many years

lthall, 2023-01-05:
> well, thank you for this insight and it is great to see that you are not among the normal people

Actually, I am one of those normal people that have been evolved in complex control systems for many years, that have made this mistake. I have seen numerous other people with a similar depth of experience make the same mistake. I have also had to do the work to rectify this mistake numerous times. Admittedly a gimbal is a relatively simple control system, maybe even simpler than I imagine.

Maybe this is the same mistake I and others have made, maybe it isn't. I don't care and it is pretty clear you don't either. So all good.

rmackay9, 2023-01-06:
OK, thanks again for the discussion above.  I've created a branch here that I think does what you guys think it should https://github.com/rmackay9/rmackay9-ardupilot/commits/gimbal-yaw-ff

I'll create a PR from it after a bit more testing.

rmackay9, 2023-01-07:
Here's a PR that I hope fixes this issue https://github.com/ArduPilot/ardupilot/pull/22604.  All feedback is more than welcome!

olliw42, 2023-01-07:
so I'm closing
hopefully a solution for plane will eventually be found too

closed 2023-01-07 by olliw42