ArduPilot/ardupilot #22572: Not using full throttle in non-airspeed takeoff

tridge, 2023-01-04:
With ARSPD_USE=0 fixed wing takeoff doesn't reach full throttle in AUTO takeoff. This is particularly a problem for catapult or bungee launches where full throttle may be needed to achieve reliable takeoff
See https://discuss.ardupilot.org/t/takeoff-throttle-is-not-at-maximum-when-airspeed-not-used-for-throttle-control/94359

With ARSPD_USE=1 we have a override in AP_TECS to give full throttle until the target airspeed is reached:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_TECS/AP_TECS.cpp#L748
there is no equivalent check in AP_TECS::_update_throttle_without_airspeed()

@priseborough any suggestions on the right approach? Overspeed can happen without an airspeed sensor of course, so just forcing full throttle for the entire takeoff isn't ideal

priseborough, 2023-01-04:
I would suggest a separate takeoff throttle parameter for the non airspeed case that would work in conjunction with the takeoff pitch angle to provide a safe airspeed.

tridge, 2023-01-04:
@priseborough thanks Paul. How would this new param combine with the pitch angle? 
I wondered if we should just have a TKOFF_THR_MIN, defaulting to zero, applied for both airspeed and non-airspeed case. That has the advantage of simplicity and not breaking existing setups.
Could you expand a bit on your suggestion?

timtuxworth, 2023-01-04:
The docs say this:
`In ArduPlane 4.0 and later, Automatic Takeoff is also a mode itself. When entered, the plane will use maximum throttle as set by the [TKOFF_THR_MAX](https://ardupilot.org/plane/docs/parameters.html#tkoff-thr-max) parameter.`

I've just been setting up a new plane based on this assumption. If this is not what is happening, is it not just a bug?

tridge, 2023-01-04:
PR to fix this here:
https://github.com/ArduPilot/ardupilot/pull/22584

PR #22584 (2023-01-15): Plane: added TKOFF_THR_MAX_T for non-airspeed takeoff
This adds TKOFF_THR_MAX_T which is the time to force maximum throttle on fixed wing takeoff without airspeed

fixes #22572
closed 2023-01-15 by tridge