ArduPilot/ardupilot #22577: should revert automate sim_vehicle in WSL2 ?

xianglunkai, 2023-01-04:
on current master branch, i think we should revert [Tools/autotest: automate sim_vehicle in WSL2]

My machine： ubuntu 18.04 in VM

SIM_VEHICLE: Using defaults from (../Tools/autotest/default_params/rover.parm)
SIM_VEHICLE: Run Rover
SIM_VEHICLE: "/home/xlk/usr_ws/ardupilot/Tools/autotest/run_in_terminal_window.sh" "Rover" "/home/xlk/usr_ws/ardupilot/build/sitl/bin/ardurover" "-S" "--model" "rover" "--speedup" "1" "--slave" "0" "--defaults" "../Tools/autotest/default_params/rover.parm" "-I0" "--home" "30.4467875,114.3963881,0.0,55.0"
RiTW: Starting Rover : /home/xlk/usr_ws/ardupilot/build/sitl/bin/ardurover -S --model rover --speedup 1 --slave 0 --defaults ../Tools/autotest/default_params/rover.parm -I0 --home 30.4467875,114.3963881,0.0,55.0
Traceback (most recent call last):
  File "/home/xlk/usr_ws/ardupilot/Tools/autotest/sim_vehicle.py", line 1593, in <module>
    start_mavproxy(cmd_opts, frame_infos)
  File "/home/xlk/usr_ws/ardupilot/Tools/autotest/sim_vehicle.py", line 831, in start_mavproxy
    wsl2_host_ip_str = wsl2_host_ip()
  File "/home/xlk/usr_ws/ardupilot/Tools/autotest/sim_vehicle.py", line 204, in wsl2_host_ip
    if not under_wsl2():
  File "/home/xlk/usr_ws/ardupilot/Tools/autotest/sim_vehicle.py", line 200, in under_wsl2
    return 'microsoft-standard-WSL2' in uname().release
AttributeError: 'tuple' object has no attribute 'release'
SIM_VEHICLE: Killing tasks

rmackay9, 2023-01-04:
FYI @magicrub @peterbarker.  Ooops.

IamPete1, 2023-01-04:
Working for me on WSL2 18.04, which python version are you running?

magicrub, 2023-01-04:
I developed that PR using python 3.10.6

magicrub, 2023-01-04:
@xianglunkai please test PR https://github.com/ArduPilot/ardupilot/pull/22586 and report good/bad on that PR so we can get this fixed ASAP

peterbarker, 2023-01-04:
https://github.com/ArduPilot/ardupilot/pull/22587 may also fix - but Tom's might be better still.

xianglunkai, 2023-01-05:
@magicrub 
I have been tested the PR #22586 , which is completely correct!
This problem has been fixed.
Thank you very much!

peterbarker, 2023-01-06:
Thanks @xianglunkai - I've merged that PR.  Apologies for the breakage.

closed 2023-01-06 by peterbarker