ArduPilot/ardupilot #22615: clang fix needed for the equivalent gcc fix here: https://github.com/ArduPilot/ardupilot/pull/22614

davidbuzz, 2023-01-09:
## Bug report

new() expects exceptions, and we compile without them, and so when any compiler optimisation level above -O0 is used, godbolt shows that some did-new-return-nullptr checks are elided by the optimisations that shouldn't be.

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

davidbuzz, 2023-01-09:
clang understands check-new but doesn't use it.

davidbuzz, 2023-01-09:
the linux boards are the ones usually compiled with clang, so this is mostly gonna afffect them.

davidbuzz, 2023-01-09:
... above my paygrade to fix clang. :-)

khancyr, 2023-01-09:
why not use C++ "smart pointer" like unique_ptr, this is the normal replacement for new() and is noexcept ?

davidbuzz, 2023-01-11:
@khancyr  - thanks , but what this issue needs is an actual fix, not random unimplemented suggestion/s.   if you have a PR that fixes this issue using 'smart pointer' that would be awesome, but there's probably a number of other ways to fix it too.

peterbarker, 2023-02-17:
> @khancyr - thanks , but what this issue needs is an actual fix, not random unimplemented suggestion/s. if you have a PR that fixes this issue using 'smart pointer' that would be awesome, but there's probably a number of other ways to fix it too.

No, that wouldn't be awesome.

I loathe the smart pointer stuff.

See https://github.com/ArduPilot/ardupilot/pull/16138

kknives, 2023-02-17:
I tried using Clang's new `-fno-new-infallible` flag, hoping that it would add the null checks for `new` back again. But no luck. `-fno-new-infallible` does nothing by itself. `-fcheck-new` is still a [FIXME on clang ](https://github.com/llvm/llvm-project/blob/main/clang/lib/StaticAnalyzer/Core/ExprEngineCXX.cpp#L967-L970).
I came up with a [small test](https://godbolt.org/z/bahrco7KT) to examine the effects these flags have. Maybe it can be useful in the future.

peterbarker, 2024-07-31:
We've moved to `std::nothrow`, which means clang++ won't optimize away the nullptr checks.

Closing this.

closed 2024-07-31 by peterbarker