ArduPilot/ardupilot #22629: Gimbal control

m98cami, 2023-01-10:
I want to send a hexadecimal message using the MAVLINK protocol in order to control a gimbal, where specifically could I modify the information in the message to include my hexadecimal string?

IamPete1, 2023-01-10:
This is not the place for questions. We do have a discord and a fourm. 

https://ardupilot.org/dev/docs/common-contact-us.html

closed 2023-01-10 by IamPete1

rmackay9, 2023-01-11:
@m98cami,

As IamPete1 says, the forum is probably a good place to ask.  Maybe ask in the [copter-4.3 forum](https://discuss.ardupilot.org/c/arducopter/copter43/167) and feel free to put "@rmackay9" in your message so I'm alerted.  I'm doing a lot of work on gimbal support these days so I think I can help.