ArduPilot/ardupilot #22657: copter flight modes not available until FRAME_CLASS set

tridge, 2023-01-14:
When first installing FRAME_CLASS won't be setup. Until it is setup the user cannot setup flight modes, which can be confusing as the link between them is not at all obvious
[image]
we should make flight modes setup work before FRAME_CLASS is set

IamPete1, 2023-01-14:
Fixed by: https://github.com/ArduPilot/MissionPlanner/pull/2972

Do you need to disconnect and reconnect or is a param re-fresh enough?

tridge, 2023-01-14:
> Do you need to disconnect and reconnect or is a param re-fresh enough?

you need to disconnect/reconnect, which is really non-obvious

I like your MissionPlanner PR, but it won't work for some builds as they override the strings (vendors often do this)

I think we should do this too: https://github.com/ArduPilot/ardupilot/pull/22658

PR #22658 (2023-01-17): Copter: allow for flight mode UI when FRAME_CLASS=0
this makes the user experience of a first time install nicer for copters/helis. At the moment the user finds the GCS is not functional, for example flight mode drop-downs don't display. This can be confusing as the user doesn't know about the link between displaying flight modes and the FRAME_CLASS
fixes #22657

Note that when FRAME_CLASS=0 we do allocate a MultiCopter matrix backend, even though there are lots of places that check for motors==nullptr
closed 2023-01-17 by rmackay9