ArduPilot/ardupilot #22662: 4.2.3: Message redefinition in .bin log

Georacer, 2023-01-15:
## Bug report

**Issue details**

I have been made aware of this issue via a couple of tickets raised in Ardupilog:
- https://github.com/Georacer/ardupilog/pull/88
- https://github.com/Georacer/ardupilog/issues/89

In the .bin logs, some messages are defined more than once by the FMT message.
Examples of such messages are:

- PRXR
- RMGH
- MON
- RSSI

Although Ardupilog has been patched to ignore the second FMT definition, the issue remains: I do not thing this is healthy behaviour from the side of Ardupilot and might signal a deeper problem.

**Version**
Users started reporting this a couple of weeks ago. One mentioned Copter 4.3.2.

**Platform**

I have been aware of **at least** Copter suffering from this.

[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
I do not know which airframe type was used.

**Hardware type**
I do not know which hardware was used.

**Logs**
A log is available here: https://github.com/Georacer/ardupilog/pull/88#issuecomment-1368286368

fagoth, 2023-01-15:
It can duplicate other records in different logs, for example 103/PIDN + 52/XKQ
My config is: Pixhawk4-bdshot at Holybro S500, airframe type X

rmackay9, 2023-01-15:
fYI @peterbarker

rmackay9, 2023-01-18:
@Georacer, @IRezkov,

Could you re-test with "latest" (aka "4.4.0-DEV").  This can be loaded directly from MP's Install Firmware screen if "Ctrl-Q" is pressed first.  After pushing Ctrl-Q the version displayed below the vehicle icons should change to "4.4.0-DEV".

Georacer, 2023-01-18:
I don't have easy access to Windows nor a FC hardware to easily create a log. Perhaps you can point me to the autotest archives with at the correct build?
Otherwise @IRezkov perhaps you can flash it and generate a log that yo or I can check.

IRezkov, 2023-01-19:
Will test at weekend

IRezkov, 2023-01-21:
Seems to be it works, test log attached.
Seen the message for the first time "PreArm: Main loop slow (231Hz < 400Hz)" during download log.
[20230121tst.zip](https://github.com/ArduPilot/ardupilot/files/10472861/20230121tst.zip)

Georacer, 2023-01-22:
Indeed, Ardupilog doesn't issue any relevant warnings with:
```
platform: 'ArduCopter'
version: 'V4.4.0-dev'
commit: '773129ba'
```

Georacer, 2023-01-22:
I consider this closed.

rmackay9, 2023-01-22:
Great, thanks very much!

closed 2023-01-22 by rmackay9

IRezkov, 2023-01-23:
THnks dear developers!