ArduPilot/ardupilot #22691: bash Tools/environment_install/install-prereqs-ubuntu.sh -y   completion   error

LouisRenWeiWei, 2023-01-19:
+ echo Adding ArduPilot Tools to environment
Adding ArduPilot Tools to environment
+ echo '##############################################'
##############################################
+++ realpath Tools/environment_install/install-prereqs-ubuntu.sh
++ dirname /home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/environment_install/install-prereqs-ubuntu.sh
+ SCRIPT_DIR=/home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/environment_install
++ realpath /home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/environment_install/../../
+ ARDUPILOT_ROOT=/home/wwren/Desktop/robot/ardupilot/ardupilot
+ [[ 1 -eq 1 ]]
+ exportline='export PATH=/opt/gcc-arm-none-eabi-10-2020-q4-major/bin:$PATH'
+ grep -Fxq 'export PATH=/opt/gcc-arm-none-eabi-10-2020-q4-major/bin:$PATH' /home/wwren/.profile
+ exportline2='export PATH=/home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/autotest:$PATH'
+ grep -Fxq 'export PATH=/home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/autotest:$PATH' /home/wwren/.profile
+ [[ '' -ne 1 ]]
+ exportline3='source /home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/completion/completion.bash'
+ grep -Fxq 'source /home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/completion/completion.bash' /home/wwren/.profile
+ maybe_prompt_user 'Add ArduPilot Bash Completion to your bash shell [N/y]?'
+ true
+ return 0
+ echo source /home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/completion/completion.bash
+ eval source /home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/completion/completion.bash
++ source /home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/completion/completion.bash
+++ $'\r'
/home/wwren/Desktop/robot/ardupilot/ardupilot/Tools/completion/completion.bash: line 2: $'\r': command not found

LouisRenWeiWei, 2023-01-19:
in unbuntu need to modify \r\n  to  \r  

sed -i 's/\r//'  completion.bash
sed -i 's/\r//' bash/_ap_autotest
ed -i 's/\r//' bash/_ap_bin
sed -i 's/\r//' bash/_sim_vehicle
sed -i 's/\r//' bash/_waf

khancyr, 2023-01-23:
hello, 
did you clone this on windows ?

maichongju, 2023-11-03:
I clone the repository in Windows and then use it with WSL. Then this happens. I believe I just need to clone the project from the WSL