ArduPilot/ardupilot #22694: Permission denied: 'ardupilot/Tools/scripts/make_intel_hex.py'

LouisRenWeiWei, 2023-01-19:
[885/887] apj_gen build/Pixhawk1/bin/arduplane.bin
[886/887] bin cleanup build/Pixhawk1/bin/arduplane.bin
[887/887] Generating bin/arduplane.hex
Waf: Leaving directory `/home/wwren/Desktop/robot/ardupilot/build/Pixhawk1'
Build failed
Traceback (most recent call last):
  File "/home/wwren/Desktop/robot/ardupilot/modules/waf/waflib/Context.py", line 363, in exec_command
    ret, out, err = Utils.run_process(cmd, kw, cargs)
  File "/home/wwren/Desktop/robot/ardupilot/modules/waf/waflib/Utils.py", line 978, in run_process
    return run_prefork_process(cmd, kwargs, cargs)
  File "/home/wwren/Desktop/robot/ardupilot/modules/waf/waflib/Utils.py", line 913, in run_prefork_process
    raise Exception(trace)
Exception: ['/home/wwren/Desktop/robot/ardupilot/Tools/scripts/make_intel_hex.py', 'bin/arduplane.bin', '../../Tools/bootloaders/Pixhawk1_bl.bin', '16']
Traceback (most recent call last):
  File "<string>", line 33, in run
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: '/home/wwren/Desktop/robot/ardupilot/Tools/scripts/make_intel_hex.py'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/wwren/Desktop/robot/ardupilot/modules/waf/waflib/Task.py", line 348, in process
    ret = self.run()
  File "<string>", line 21, in f
  File "/home/wwren/Desktop/robot/ardupilot/modules/waf/waflib/Task.py", line 331, in exec_command
    return self.generator.bld.exec_command(cmd, **kw)
  File "/home/wwren/Desktop/robot/ardupilot/modules/waf/waflib/Context.py", line 365, in exec_command
    raise Errors.WafError('Execution failure: %s' % str(e), ex=e)
waflib.Errors.WafError: Execution failure: ['/home/wwren/Desktop/robot/ardupilot/Tools/scripts/make_intel_hex.py', 'bin/arduplane.bin', '../../Tools/bootloaders/Pixhawk1_bl.bin', '16']
Traceback (most recent call last):
  File "<string>", line 33, in run
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: '/home/wwren/Desktop/robot/ardupilot/Tools/scripts/make_intel_hex.py'

peterbarker, 2024-07-31:
@LouisRenWeiWei did you ever resolve this?

The only thing I can think of is that the filesystem was mounted with the no-execute option.  The Python script definitely was executable at that point in time.

lashVN, 2024-08-02:
Hi @peterbarker , I actually just got the same build error (from ./waf plane) when trying to build a branch forked from AP 4.0. Did you run into this when using an old branch as well? All I can find related online is [this](https://discuss.ardupilot.org/t/install-ardupilot4-0-on-ubuntu20-04-and-compile-error/79108/13) thread implying it could be a python 2 vs 3 issue.

peterbarker, 2024-08-03:
> Hi @peterbarker , I actually just got the same build error (from ./waf plane) when trying to build a branch forked from AP 4.0. Did you run into this when using an old branch as well? All I can find [...]

Could you run some experiments, please?

PATH="/home/wwren/Desktop/robot/ardupilot/Tools/scripts/make_intel_hex.py"
`ls -l "$PATH"`
`head -1 "$PATH"`
`python2 "$PATH"`
`python3 "$PATH"`

4.0 we were part-way between Pythons, so I wouldn't be at all surprised to know there's a problem.  That version is ancient and not really supported by the team any more; the Py2/Py3 transition makes it really rather unpleasant.  Perhaps you could look at forward-porting your branch :-)

lashVN, 2024-08-03:
@peterbarker 
> Could you run some experiments, please?

Of course I had to modify "PATH" to be "PATH1" to avoid messing up the PATH env variable, but I ended up with the following:

```
>username@DESKTOP-B92RUDE:~/ardupilot_shane$ PATH1="/home/username/ardupilot_shane/Tools/scripts/make_intel_hex.py"
>username@DESKTOP-B92RUDE:~/ardupilot_shane$ ls -l "$PATH1"
-rw-r--r-- 1 username username 1315 Aug  1 19:34 /home/username/ardupilot_shane/Tools/scripts/make_intel_hex.py
>username@DESKTOP-B92RUDE:~/ardupilot_shane$ python2 "$PATH1"

Command 'python2' not found, but can be installed with:

sudo apt install python2

>username@DESKTOP-B92RUDE:~/ardupilot_shane$ python3 "$PATH1"
Usage: make_intel_hex.py BINFILE BOOTLOADER RESERVE_KB
```

Of course I don't have python2 installed which is likely related (I can try with python2 to see how it works), and it looks like I need some arguments to fake to make_intel_hex. I tried tracing down the call stack and couldn't readily find what arguments are supposed to go to it.

peterbarker, 2024-09-04:
> Of course I had to modify "PATH" to be "PATH1" to avoid messing up the PATH env variable, but I ended up with the following:

Whoops.  My bad :-)

> >username@DESKTOP-B92RUDE:~/ardupilot_shane$ PATH1="/home/username/ardupilot_shane/Tools/scripts/make_intel_hex.py"
> >username@DESKTOP-B92RUDE:~/ardupilot_shane$ ls -l "$PATH1"
> -rw-r--r-- 1 username username 1315 Aug  1 19:34 /home/username/ardupilot_shane/Tools/scripts/make_intel_hex.py

Yep, that's a problem.  That file should be executable.  Strangely, it was *created* executable in 2018, and I don't think it's changed since!

So something has stripped execute permissions from that file for you locally, I think.  Rectify it with `chmod +x home/username/ardupilot_shane/Tools/scripts/make_intel_hex.py` - 'though if you're not sure how that happened I wouldn't be surprised if there were more issues like it!

> >username@DESKTOP-B92RUDE:~/ardupilot_shane$ python2 "$PATH1"
> 
> Command 'python2' not found, but can be installed with:
> 
> sudo apt install python2
> 
> >username@DESKTOP-B92RUDE:~/ardupilot_shane$ python3 "$PATH1"
> Usage: make_intel_hex.py BINFILE BOOTLOADER RESERVE_KB
> ```

Yep, so just an executable-bit-permission problem at this point.

Once you've added the execute bit, try just running it as a command with no arguments; it may or may not find a Python it can use... let's find out...

> Of course I don't have python2 installed which is likely related (I can try with python2 to see how it works), and it looks like I need some arguments to fake to make_intel_hex. I tried tracing down the call stack and couldn't readily find what arguments are supposed to go to it.

If it works with Python3 - don't argue :-)

> > Perhaps you could look at forward-porting your branch :-)
> 
> That's something being looked into by our end customer. Unfortunately it's not my fork such that I have control over that effort!

Not an unusual situation!  Do point out to the end customer that modern hardware is vastly better than that supported in ArduPilot 4.0.x - new IMUs, new GPSs, ....

> I also didn't mean to turn this thread into you supporting my attempt to build an old fork; I went to resolve my issue and noticed you'd just marked the necessity to resolve a similar issue, so was hoping I'd be able to help in that effort.

No worries.  Let's see if this gets you going :-)

peterbarker, 2024-11-24:
Ping @lashVN - did you get past this problem?

Did you work out what stripped permissions?