ArduPilot/ardupilot #22704: System Identification Code Missing for Simulink Model in Tools

n-shubham, 2023-01-21:
System Identification Code Missing for Simulink Model in Tools 

Currently, the simulink model seems to be using dummy models for simulation.
I see that there is some code to load preexisting models in `sid_sim_init.m` but no code to generate these models. Can we add these functions? 

Further, would appreciate inputs on how to use the simulation to tune a copter's parameters.

@fbredeme @amilcarlucas

amilcarlucas, 2023-01-23:
Yes, it uses dummy transferfunctions, because you have not yet provided it wit the required data to perform system identification.

Once you do testfligjhts and provide that data, the sripts will replace the simplified transfer functions with accurate transfer functions from YOUR copter.
You can, should and must add that data.

https://github.com/fbredeme/ardupilot_wiki/blob/pr-system-id-improvements/copter/source/docs/systemid-mode-operation.rst
https://github.com/fbredeme/ardupilot_wiki/blob/pr-system-id-improvements/copter/source/docs/systemid-mode-ctrl-optimization.rst

fbredeme, 2023-01-23:
@amilcarlucas To be precise, the script does not yet contain our system identification functionality. For now, you need to identify the Copter by yourself @n-shubham