ArduPilot/ardupilot #22721: VTX_MAX_POWER upper bound too low

cbf123, 2023-01-23:
There are 2000mW VTXs out there now, so it seems wrong to limit VTX_MAX_POWER to 1000.

supercrossed, 2023-04-19:
Agreed. I'm trying to setup the Rush solo max - 2.5w and can only get it to 1000mw via smart audio. Setting it to 2500mw will actually set it to 25mW.

cbf123, 2024-04-16:
Seems to still be an issue in AP_VideoTX.cpp as upper bound is still 1000

    // @Range: 25 1000
    AP_GROUPINFO("MAX_POWER", 7, AP_VideoTX, _max_power_mw, 800),

DenysZaytsev, 2024-09-29:
I have AKK Ultra Long Range VTX which is up to 3000mW, but with Ardupilot it can be set only to 800mW

How to fix that? This is now crucial!

menschel, 2024-12-06:
The issue is this table which tries to handle tramp (mW), smart audio V1(dac), V2(mW) and V2.1(dbm) values
```
AP_VideoTX::PowerLevel AP_VideoTX::_power_levels[VTX_MAX_POWER_LEVELS] = {
    // level, mw, dbm, dac
    { 0xFF,  0,    0, 0    }, // only in SA 2.1
    { 0,    25,   14, 7    },
    { 0x11, 100,  20, 0xFF }, // only in SA 2.1
    { 1,    200,  23, 16   },
    { 0x12, 400,  26, 0xFF }, // only in SA 2.1
    { 2,    500,  27, 25   },
    { 0x12, 600,  28, 0xFF }, // Tramp lies above power levels and always returns 25/100/200/400/600
    { 3,    800,  29, 40   },
    { 0x13, 1000, 30, 0xFF }, // only in SA 2.1
    { 0xFF, 0,    0,  0XFF, PowerActive::Inactive }  // slot reserved for a custom power level
};
```
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_VideoTX/AP_VideoTX.cpp#L110

Betaflight has this simplified to Label / Value -tuples, no logic behind it.

DenysZaytsev, 2024-12-06:
That's weird that Betaflight is just supporting any VTX, you just put the values in json and import (or manually type in the UI), but instead Ardupilot doesn't support VTX with > 1000mW of power and no swithing between different power levels for different VTXs (i.e. Ardupilot can only switch between 0/25/500/600/800/1000, but different VTX may support completely different power lewels: 2000mw/3000mw, etc..)

menschel, 2024-12-06:
I assume you have a SA2.0 VTX then.
The thing is, that this table has nothing above 1000mW, so it won't be taken as a value.
It is a concept issue. Betaflight just maps Value and Label, means it does not care about the underlying protocol and mechanisms.

ccc321q, 2026-09-11:
I am running into this problem at the moment, my VTX does not seem to take the standard 25/100/200/400/600 inputs, but the actual mA values. I need 25/400/800/1500/2500. It is an ULTRA2500 VTX 2.5W using IRC Tramp and I can only go up to 800mW.

ccc321q, 2026-09-12:
Fixed in #34379