ArduPilot/ardupilot #22724: text to speech trigger for STATUSTEXT

tridge, 2023-01-23:
I'd like a way for the autopilot to send a STATUSTEXT which it wants the GCS to say out loud as a text to speech. This came up for the aerobatics, but I think it would be generally useful in lua scripts
ideas:
 - add a new MAV_SEVERITY_ANNOUNCE level which would indicate the GCS should process as text to speech
 - use a string convention, say an exclamation mark "!Hello world" and GCS would display

Adding MAV_SEVERITY_ANNOUNCE is very clear, but we'd need to see what existing GCS do with it.

timtuxworth, 2023-01-23:
I totally agree that current use of "Critical" messages just so that a ground station will announce it leads to confusion. This needs a solution. However using the SEVERITY imposes some limitations, for example sometimes you want an INFO to be announced (or not), sometimes something is Warning, but does not need to be announced. We don't really want to spam GS because now we need two messages, one for the real severity and one for the announcement. It's also not obvious what an existing older groundstation will do with a message containing what it thinks is an invalid severity.

Perhaps an extension field would be better - could be a flag "ANNOUNCE"? This way everything  will continue to work with existing Ground Stations (which will not understand a new MAV_SEVERITY), but when upgraded will also understand the "Announce" extension field.

tridge, 2023-01-23:
> It's also not obvious what an existing older groundstation will do with a message containing what it thinks is an invalid severity.

that's easy enough to test, I suspect they would treat like MAV_SEVERITY_DEBUG

muramura, 2023-01-24:
QGC is a high-level message above notice, and mp is a high-level message above warning.  The latest mp can specify the voice notification level.

hamishwillee, 2023-01-24:
Isn't this what https://mavlink.io/en/messages/common.html#MAV_SEVERITY_ALERT is for? Or is this more like "instructions" that makes sense for user to hear?

- Should it be logged?
- Do we really need more sound coming out of a GCS?/Will this clash with actual important warnings?
- Assume that a GCS can ignore this if it wants. If so, treat as "INFO" severity?

Have no strong opinion other than:
- If this is wanted, please add upstream PR so I can get broader testing/discussion.
- Could this be in [EVENT interface](https://mavlink.io/en/services/events.html) - i.e. does it require reliable delivery etc.

muramura, 2023-01-25:
I would like to specify the audio notification regardless of the minimum gcs audio notification level.  The alert specification will be notified by voice at the lowest level specification.  I am specifying notice with the intention of making an audio notification.  I would prefer that this announcement not be logged in the flight log.