ArduPilot/ardupilot #22725: Autotest script can blow away local changes

sypaq-nexton, 2023-01-23:
## Bug report

**Issue details**

Running the `autotest.py` script can blow away local code changes:
* If no test is specified, the `autotest.py` script will run all available tests.
* One of the tests is "Build Binaries", which calls the `build_binaries.py` script.
* The `build_binaries.py` script does a *forced checkout* of the `master` branch, blowing away any local changes.

**Version**
All

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

tridge, 2023-01-24:
> If no test is specified, the `autotest.py` script will run all available tests.

it doesn't any more, it gives this:
```
Steps must be supplied; try --list and/or --list-subtests or --help
```

sypaq-nexton, 2023-01-24:
Yep, it's fixed on `master`. I've closed PR #22726.

However, the fix hasn't made it into any of the platform branches.

peterbarker, 2024-07-31:
> Yep, it's fixed on `master`. I've closed PR #22726.
> 
> However, the fix hasn't made it into any of the platform branches. Happy to close this if you don't think that's a problem though.

4.5.x will have the fix in now, so closing this one.

It was a really sharp edge, wasn't it?

closed 2024-07-31 by peterbarker