ArduPilot/ardupilot #22769: Rangefinder VL53L1X has a delta in measurement of 8cm

olif63, 2023-01-28:
## Bug report

**Issue details**
I am not able to take off my 3 inch copter in loiter mode because I get no reading from rangefinder VL53L1X when standing on ground. Ground clear is 5cm, sensor is working from 4 cm on, so I should get a reading of 1cm.
The fact is that the sensor starts reading from a height of 13cm on. It shows 1cm then.
This behavior is confirmed by ppoirier. The VL53L1X has a delta of 8cm, the VL53L0X measures the right height.
I tested the sensor on the drone against the same sensor running with a common arduino script on arduino uno.

**Version**
arducopter 4.3.2

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Qadrocopter

**Hardware type**
Matek H743 mini

**Logs**
_Please provide a link to any relevant logs that show the issue_
[image]

shivam969619, 2023-02-04:
Calibrate the sensor to see if this improves accuracy.
Check the wiring and connections to ensure that there are no issues with the electrical setup.
Try using the VL53L0X sensor instead of the VL53L1X to see if it measures the right height.
Check if there are any known issues with the VL53L1X sensor and the ArduCopter platform, and if so, see if there are any solutions or workarounds available.
Reach out to the manufacturer of the VL53L1X sensor or to the ArduCopter community for further assistance.

olif63, 2023-02-04:
My System is running with VL53L0X but it is working only up to 120cm. The sensor is o.k. as it is working on Arduino as expected.
The error is confirmed as you can read here https://discuss.ardupilot.org/t/vl53l1x-distance-offset/95954

muramura, 2023-02-05:
This driver does not determine the minimum distance.
This device has a minimum distance of 4 cm.
It does not treat distances shorter than 4 cm as invalid values.
Furthermore, the distance is the average of the accumulated distances.
The accumulation of illegal values results in an illegal average value.
I believe that the indication of 1 cm is using an illegal value and making it an average value.

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_VL53L1X.cpp#L559

DataSheet:
https://www.pololu.com/file/0J1506/VL53L1X.pdf