ArduPilot/ardupilot #22779: Cube Orange unable to detect Barometer with Ardupilot Firmware

abhisheeekkk, 2023-01-30:
## Bug report

**Issue details**

Hello All, 
               We were using cube orange for Arducopter and Arduplane firmware and it was working properly, but today when I am trying to flash the firmware of ardupilot it is throwing this error :

```
30.11.2021 13:52:20 : T Failed
30.11.2021 13:52:20 : Config error: Board Validation $CHECK_BARO1_PRESEN
30.11.2021 13:52:20 : Config Error: fix problem then reboot
```
[image]

We tried all the firmwares of Arduplane and Arducopter (i.e. from V4.0.0 to V4.3.3) but are unable to resolve this issue, Because of this issue we are unable to move forward because it won't let us do the calibration of sensors or radio. I tried flashing the firmware of PX4 and that doesn't throw this error, but Ardupilot firmwar is required in our application, Please if you guys have any idea what could be the root cause of the issue please let us know, Thanks in advance!

I also have opened issue at discussion forum - https://discuss.ardupilot.org/t/cube-orange-unable-to-detect-barometer/96115

**Version**
V4.0.0 to V4.3.3 of Arduplane and Arducopter

**Platform**
[  ] All
[  ] AntennaTracker
[ - ] Copter
[-  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad and Octa

**Hardware type**
Cube Orange

**Logs**
```
30.11.2021 13:52:20 : T Failed
30.11.2021 13:52:20 : Config error: Board Validation $CHECK_BARO1_PRESEN
30.11.2021 13:52:20 : Config Error: fix problem then reboot
```

rishabsingh3003, 2023-02-01:
Probably something that requires the attention of CubePilot? Have you tried posting on CubePilot forum?

rmackay9, 2023-02-01:
I think this is most likely a hardware issue and in any case it is a support issue so I will go ahead and close this issue.  Txs for the report and sorry to be strict.  If we're not then our to-do list (which is what this issues list is) just turns into another support forum.

closed 2023-02-01 by rmackay9